nordicsemi/IOS-DFU-Library #1: Pod install error

sdespont, 2016-05-08:
Hello,

I am trying to install the framework using podfile as explained in the documentation, but I am getting this error : 

`[!] Invalid `Podfile` file: syntax error, unexpected keyword_end, expecting end-of-input.`

My podfile is:
```
use_frameworks!
    pod 'iOSDFULibrary'
end
```

Any idea?

mostafaberg, 2016-05-08:
Can you attach your `Podfile` here ? sometimes quotes display differently and are actually broken, you need to make sure they're actual single quotes and not the curled ones

sdespont, 2016-05-08:
Sure, here is the zipped file :
[Podfile.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/254102/Podfile.zip)

This file has been created with TextWrangler in order to avoid "quotting" problem with TextEdit

mostafaberg, 2016-05-08:
Oh, my bad, just noticed a tiny mistake in the documentation, we missed out the `target` setting, will update the docs tomorrow !

your issue is that you're missing the `target 'myApp' do` in the beginning, so the Podfile looks like this:

```
use_frameworks!
target 'MyApp' do
  pod 'iOSDFULibrary'
end
```

Sorry about that mistake, let me know if this resolves your issue :+1:

sdespont, 2016-05-08:
Thanks, first time using CocoaPods for me, sorry for the dumb question.

Now, the error is the following : 

```
Analyzing dependencies
[!] Unable to find a specification for `iOSDFULibrary`
```

Can you help me or give me a hint?

mostafaberg, 2016-05-08:
No worries at all, if it can't find a specification for your app that means you probably have a deployment target we don't support, what's your deployment target in XCode ?

sdespont, 2016-05-08:
IOS 9.2

mostafaberg, 2016-05-08:
What happens if you try to run this in your terminal ?

`pod try iOSDFULibrary`

are you able to build and run the example ?

sdespont, 2016-05-08:
Another error message :

```
Trying iOSDFULibrary
Performing CocoaPods Installation

[!] Invalid `Podfile` file: [!] Unsupported options `{:exclusive=>true}` for target `iOSDFULibrary_Example`..

 #  from /private/var/folders/_h/vmjtkgp517q_4vwbmf59fgvc0000gr/T/CocoaPods/Try/iOSDFULibrary/Example/Podfile:3
 #  -------------------------------------------
 #  
 >  target 'iOSDFULibrary_Example', :exclusive => true do
 #    pod 'iOSDFULibrary', :path => '../'
 #  -------------------------------------------

```

mostafaberg, 2016-05-08:
Well that's interesting, I need to get a hold on a Mac to test this out, I haven't seen things go that way before, so I'll have to wait till tomorrow morning before I can be of any help.

mostafaberg, 2016-05-09:
Just made it to work, I cannot seem to reproduce the issue, can you try reinstalling Cocoapods perhaps ? could be bad caches maybe ?
here's the output of my `pod try iOSDFULibrary`

```
Trying iOSDFULibrary
Performing CocoaPods Installation

CocoaPods 1.0.0.rc.2 is available.
To update use: `gem install cocoapods --pre`
[!] This is a test version we'd love you to try.

For more information see http://blog.cocoapods.org
and the CHANGELOG for this version http://git.io/BaH8pQ.

Fetching podspec for `iOSDFULibrary` from `../`
Installing EVReflection 2.29.0 (was 2.24.0)
Using Zip (0.3.4)
Installing iOSDFULibrary 0.1.5 (was 0.1.1)
Sending stats
Pod installation complete! There is 1 dependency from the Podfile and 3 total pods installed.
Opening '/private/var/folders/7r/nxyqj1lx4k93crsknywzn5880000gn/T/CocoaPods/Try/iOSDFULibrary/Example/iOSDFULibrary.xcworkspace'
```

mostafaberg, 2016-05-09:
I just made it to work and cannot seem to reproduce the issue, here's the output of a newly created project, made that `Podfile` in the project's root : 

```
use_frameworks!
target 'dfuTesting' do
	pod 'iOSDFULibrary', '~> 0.1'
end
```

and here's the output of `pod install`

```
CocoaPods 1.0.0.rc.2 is available.
To update use: `gem install cocoapods --pre`
[!] This is a test version we'd love you to try.

For more information see http://blog.cocoapods.org
and the CHANGELOG for this version http://git.io/BaH8pQ.

Analyzing dependencies
Downloading dependencies
Installing EVReflection (2.29.0)
Installing Zip (0.3.4)
Installing iOSDFULibrary (0.1.5)
Generating Pods project
Integrating client project

[!] Please close any current Xcode sessions and use `dfuTesting.xcworkspace` for this project from now on.
Sending stats
Pod installation complete! There is 1 dependency from the Podfile and 3 total pods installed.
```

Also here's the output of `pod try iOSDFULibrary` : 

```
Trying iOSDFULibrary
Performing CocoaPods Installation

CocoaPods 1.0.0.rc.2 is available.
To update use: `gem install cocoapods --pre`
[!] This is a test version we'd love you to try.

For more information see http://blog.cocoapods.org
and the CHANGELOG for this version http://git.io/BaH8pQ.

Fetching podspec for `iOSDFULibrary` from `../`
Installing EVReflection 2.29.0 (was 2.24.0)
Using Zip (0.3.4)
Installing iOSDFULibrary 0.1.5 (was 0.1.1)
Sending stats
Pod installation complete! There is 1 dependency from the Podfile and 3 total pods installed.
Opening '/private/var/folders/7r/nxyqj1lx4k93crsknywzn5880000gn/T/CocoaPods/Try/iOSDFULibrary/Example/iOSDFULibrary.xcworkspace'
```

Maybe if you try to clear the caches ? try to `rm -rf ~/Library/Caches/CocoaPods` and see if that helps ?

sdespont, 2016-05-09:
Thanks for your tests. Your assumptions was correct, clearing the cache permits to run successfully the command `pod install` 👍

mostafaberg, 2016-05-09:
That's awesome !, Cocoapods can be tricky sometimes 👍  glad to hear it's resolved now, have a great day

closed 2016-05-09 by mostafaberg