nordicsemi/IOS-DFU-Library #3: Compiling error with IOS-Pods-DFU-Library installed

sdespont, 2016-05-09:
After install your POD in my XCode project, I get the following error when compiling : 

```
ld: warning: directory not found for option '-F/Users/Despont/Library/Developer/Xcode/DerivedData/SHR-cebhiiybhzbjpgefaonqtocowutd/Build/Products/Debug-iphoneos/EVReflection'
ld: warning: directory not found for option '-F/Users/Despont/Library/Developer/Xcode/DerivedData/SHR-cebhiiybhzbjpgefaonqtocowutd/Build/Products/Debug-iphoneos/Zip'
ld: warning: directory not found for option '-F/Users/Despont/Library/Developer/Xcode/DerivedData/SHR-cebhiiybhzbjpgefaonqtocowutd/Build/Products/Debug-iphoneos/iOSDFULibrary'
ld: framework not found EVReflection
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

mostafaberg, 2016-05-09:
Are you working from .xcworkspace or from .xcodeproj ? you need to open and work in the workspace when using cocoapods

sdespont, 2016-05-09:
That's it! Thank you for your really quick answer.

closed 2016-05-09 by mostafaberg