nordicsemi/IOS-DFU-Library #4: what is  the dat file?

fengguixian, 2016-05-11:
When i want to send the hex file to the fw, but faild...

Error 103: The init packet is required by the target device

philips77, 2016-05-11:
Hi @fengguixian, as the message explains, since SDK 7.1 the DFU bootloader requires an Init packet, in addition to the firmware file. Instead of using just a HEX file, now you have to use a ZIP file with a bin file (stripped data from the hex), a dat file (the init packet file) and manifest.json file. Such a ZIP file, called a Distribution Packet, can be automatically created using the **nrfutil** tool. Please, read more [here](https://github.com/NordicSemiconductor/nRF-Master-Control-Panel/tree/master/init%20packet%20handling). The **nrfutil** can be downloaded as an executable in PC version of MCP 3.8+, or [here](https://github.com/NordicSemiconductor/pc-nrfutil).

fengguixian, 2016-05-13:
Thank you for your soon replying. After I readed the pdf file over and over, luckily, I can generate the dat file and bin file with the nrfutil tool. But i failed again...
Error 104: Init packet with 2-byte CRC supported. Extended init packet found.

philips77, 2016-05-13:
This is weird.. I have to look into it. 
What DFU bootloader you are using? From which SDK version?
The first message you got, error 103, should be only shown if there is a DFU Version characteristics (sdk 7.1 or newer).
The 104 error, on the other hand - before 7.1.
Do you have this characteristic? What value does it return?

To generate an init packet with only 2-byte CRC edit the .dat file in the zip and delete all bytes except the last two. But this should only be required if you have a bootloader from sdk before 7.1, no DFU Version char, and want to send a SoftDevice and bootloader update.

fengguixian, 2016-05-13:
My SDK is 7.2, below is the error message log for your reference. 

[image]
)

philips77, 2016-05-13:
Hi, so I looked at your listing and the code and here are my conclusions:
You don't have the *DFU Version* characteristic in the *DFU Service*. It was added in SDK 7.1. Are you sure you have the bootloader from 7.2? 
Otherwise, it would write "Reading DFU Version characteristic... [...]" after "DFU characteristics discovered". Instead it started to enable notifications.
At that moment it knows that you have an older bootloader, that does not understand the Extended Init Packet.
However, when it comes to sending the (optional before SDK 7.1) init packet, it checks its length which is longer than 2 bytes (the DFU bootloader back then supported only the CRC check and the init packet could have had 2 bytes) and notifies you with an error.

Solutions to this problem are two:

1. Make sure your DFU Bootloader is from SDK 7.1+ and there is a DFU Version characteristic after you connect to it. You may verify it for example using nRF Master Control Panel on iOS or Android.
Then it should work with the init file you have. The DFU Version char must return 0x0100 in the application mode and 0x0[5,6,7,8]00 in bootloader mode. Depending on the value returned the library knows what to do.
2. Cut your generated Init packet (the dat file in the ZIP) and leave only the last 2 bytes, as I wrote earlier.

Another thing is why you got the 103 error at the first place. I think, when you were testing back then, you really had the bootloader from SDK 7.2 but later you changed it to an older version. Can I be correct? Otherwise, it's not possible, in my opinion, to get both 103 and 104 using the same bootloader.

mostafaberg, 2016-05-19:
Hi @fengguixian !, have you managed to get the issue resolved ? can you confirm this still happens in  release `0.1.8` ?

fengguixian, 2016-05-23:
@mostafaberg I am sorry to read your comment so late, I don't resolve this issue yet,  I am confused for this issue now. I used the **nRF Toolbox** and put my ZIP file into it, and it will upgrade succed, but if I use this lib in my own test app, it faild. Refer to @philips77's suggestions, I must make sure the fw is right. So I am waiting the message of th fw programer. If you have some good suggestions , it is very kind to tell me, thank you very much.

mostafaberg, 2016-05-23:
No worries, so as you're saying is that it works well with the nRF Toolbox app ? that means either we have a bug in this library or there's something wrong with your app/firmware, is it possible to ask for the firmware file so I can try to reproduce the bug here ? or if you can give me access to your repo I can test it and do a pull request with the fix if i manage to fix it ? what board are you trying to flash btw ?

fengguixian, 2016-05-23:
We used our own board with nRF51822QFAAH0  on it. Actually before, we have ever used this board with nRF51822QFAAG2 and used SDK 5.1 to flash successfully by our own test APP. We did change 51822 to latest version H0 and updated SDK from 5.1 to 7.2. But now, we encountered this bug and had no idea to resolve it. I could send the firmware to you for trying. 
[application.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/276965/application.zip)

mostafaberg, 2016-05-23:
perfect, i'll take a look into that and let you know my findings, if there are any bugs in the DFU Library i'll push a fix asap, if not i'll let you know what the issue is, thanks !

mostafaberg, 2016-05-23:
Ok just did a quick test and it seems to work as intended here, using Xcode `7.3.1` and the board is a `PCA10040` with a `N52832QFAABB1601AB` chip, Here's my console output:

```
Verbose: Connecting to DfuTarg...
Debug: centralManager.connectPeripheral(peripheral, options:nil)
Debug: [Callback] Central Manager did connect peripheral
Info: Connected to DfuTarg
Verbose: Discovering services...
Debug: periphera.discoverServices(nil)
Info: Services discovered
Verbose: DFU Service found
Verbose: Discovering characteristics in DFU Service...
Debug: peripheral.discoverCharacteristics(nil, forService:DFUService)
Info: DFU characteristics discovered
Verbose: Reading DFU Version number...
Debug: peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)
Info: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):0800
Application: Version number read: 0.8
Verbose: Enabling notifiactions for 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.setNotifyValue(true, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123)
Verbose: Notifications enabled for 00001534-1212-EFDE-1523-785FEABCD123
Application: DFU Control Point notifications enabled
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x0104, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
Verbose: Writing image sizes (0b, 0b, 46724b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x000000000000000084b60000, forCharacteristic: 00001534-1212-EFDE-1523-785FEABCD123, type: WithoutResponse)
Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Application: Start DFU (Op Code = 1, Upload Mode = 4) request sent
Info: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100101
Application: Response (Op Code = 1, Status = 1) received
Application: Writing Initialize DFU Parameters...
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x0200, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
Verbose: Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0xffffffffffffffff0100feff6d87, forCharacteristic: 00001534-1212-EFDE-1523-785FEABCD123, type: WithoutResponse)
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x0201, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Info: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100201
Application: Response (Op Code = 2, Status = 1) received
Application: Initialize DFU Parameters completed
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x080c00, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Application: Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x03, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Application: Receive Firmware Image (Op Code = 3) request sent
Application: Uploading firmware...
Verbose: Sending firmware DFU Packet characteristic...
Info: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100301
Application: Response (Op Code = 3, Status = 1) received
Application: Upload completed in 18.12 seconds
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x04, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Application: Validate Firmware (Op Code = 4) request sent
Info: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100401
Application: Response (Op Code = 4, Status = 1) received
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x05, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
Info: Data written to 00001531-1212-EFDE-1523-785FEABCD123
Application: Activate and Reset (Op Code = 5) request sent
Debug: [Callback] Central Manager did disconnect peripheral
Info: Disconnected by the remote device
```

So to me it seems the library is doing it's job, I won't be able to reproduce your issue without knowing more details about your code and how you're attempting to run a DFU, if this is a private repo that you can't share here please feel free to email us and we can take that offline !, let me know if that helps

fengguixian, 2016-05-26:
@mostafaberg Thank for your help. Now I found a strange phenomenon when I do my test again and again. After I make my device enter the dfu, and use the lib for it, but it faild with the error: DFU Service not found. But after I reopen the BLE for the system,  tried it again, it would find the dfu service and work. Why?

mostafaberg, 2016-06-01:
Could that be an issue with your device perhaps ? when that happens again can you try checking if it is broadcasting a DFU  service from another phone ?

philips77, 2016-06-01:
Hi, the iOS is caching the device services unless:
1. the device is not bonded and the Service Changed characteristic is not present
2. the device is bonded and a Service Changed indication is received
To force clearing the cache you need to restart the Bluetooth adapter.

fengguixian, 2016-06-01:
@philips77 @mostafaberg Good news, this issue has been solved just now. Thank you very much.

mostafaberg, 2016-06-01:
Perfrct,  that's good to hear. Let us know if you encounter more issues!

closed 2016-06-01 by mostafaberg