nordicsemi/IOS-DFU-Library #6: Is there an option to forward CBCentralManager Delegate method?

klemenzagar91, 2016-06-16:
Hi, while in the DFU process, I want to know if user by acident turn off bluetooth, so I can change UI accordingly. Currently the delegate is used inside your library, so I can't track in which state is CBCentralManager.

```
func centralManagerDidUpdateState(central: CBCentralManager) {
        logCentralManagerState(central.state)
    }
```

This line inside your library could be used to forward this info outside your library. Thanks.

philips77, 2016-06-17:
Very good point. 
And we should also rethink reverting original CentralManager delegate.

klemenzagar91, 2016-06-17:
Great. Thanks for fast reply. 

And one more thing. When you make changes to DFU Library, do you also update this one used for cocoapods. I hope they are both in sync.

mostafaberg, 2016-06-17:
@philips77  +1, Will be on that fix soon
@klemenzagar91 yes they are both going to be in sync and released at the same time !, will close this ticket when the new update is out so you'll be notified

huguesbr, 2016-09-28:
Can you not have multiple `CBCentralManager` in order to address this issue? One that you pass to the `DFUServiceInitiator` and one that you keep for yourself? You will consume a little extra  ressources but if it's just to handle state, I think it should be all good...

mostafaberg, 2016-10-04:
Will evaluate the possibility of forwarding CentralManager delegate methods to the user's application for more granular control, will keep this open until the next release

casey-pax, 2017-01-27:
Would love to get the delegate restored! +1 Vote for that :)

mostafaberg, 2017-01-30:
@casey-pax Ok i guess we got more than enough requests to get that implemented !, will try go get the feature out asap, can't give an exact ETA but it'll probably be within the next 1 or 2 updates hopefully !

realli, 2017-06-30:
what's the status of this issue? looking forward to this feature

Looveh, 2017-09-14:
I was able to successfully work around this by resetting the delegate of the `CBCentralManager` instance after the DFU process exited (success or error) to the delegate instance used before DFU.

zurichrich, 2018-03-16:
did the central delegate restoring get implemented please?

philips77, 2018-03-16:
It's not yet implemented.

zurichrich, 2018-03-16:
Hi - Do you know of a swift solution to hand the delegate back to central after DFU is complete?

philips77, 2018-03-16:
Just restore it to what it was before DFU in completed, error or abort handlers.

zurichrich, 2018-03-17:
central.delegate = self doesnt do it. What will restore it?

philips77, 2018-10-05:
This will be invalid when #228 gets merged. Targeted in 4.2 release.

philips77, 2018-10-05:
Fixed in https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/releases/tag/4.2.0

closed 2018-10-05 by philips77