nordicsemi/IOS-DFU-Library #9: When killing the app during a dfu upload, the upload gives an invalid state error

foobar8675, 2016-06-24:
I am getting an error where I start doing a DFU upload of an application firmware, then kill the toolbox app, and then try to upload the firmware again. I always get an invalid state error.

Is it possible I need to send an `abort` to the DFUPeripheral before sending a `start`?

philips77, 2016-06-24:
I think it's related to https://github.com/NordicSemiconductor/nRF-Master-Control-Panel/issues/16

foobar8675, 2016-07-01:
hi @philips77 i didn't realize that repo above was only for android. do you know when there might be a fix on iOS i can test with?

philips77, 2016-07-01:
Hi, I guess some time early next week, but we'll see. @mostafaberg

mostafaberg, 2016-07-01:
I'm on it, I'll check if I can do a good fix on Monday !, can't promise a lot as it's not a very high priority issue but we will see next week 👍

mostafaberg, 2016-07-04:
Ok i took a look and already started working on a fix, will have something out during this week!

foobar8675, 2016-07-04:
Thank you so much @mostafaberg

mostafaberg, 2016-07-05:
No worries!,
I've just resolved the issue now and in the process of publishing the new Pod and it will be available soon, the version number you're looking for is 0.1.9 build 9, let me know if you have any other issues!

closed 2016-07-05 by mostafaberg

foobar8675, 2016-07-05:
@mostafaberg since the pod repo has not updated yet, can you create a tag on github so i can just reference the tag for testing purposes?

mostafaberg, 2016-07-05:
It's already tagged 0.1.9, you can check it out here:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/releases/tag/0.1.9

foobar8675, 2016-07-06:
Testing now, seems great.

mostafaberg, 2016-07-07:
Great news !, thanks for the feedback