nordicsemi/IOS-DFU-Library #10: Can characteristic 0x1534 be used to store firmware version

lilliesAndRoses, 2016-06-27:
Hi,
The library uses a DFU version characteristic 0x1534. The library infers certain states based on particular values of this service. Given this, I gather that this characteristic can not be used to store the actual firmware version. Am I correct? If so, is there a way to expose the firmware version as a part of DFU Service itself?

philips77, 2016-06-27:
Hi, no. And it may not be that good idea after all, as there is a security issue with it. An attacker may know that some devices are vulnerable to some kinds of attacks.

lilliesAndRoses, 2016-06-27:
So there should be a separate way of finding the firmware version to decide if upgrade is needed or not.

philips77, 2016-06-27:
Yes, this is how I see it. But please ask your question on the DevZone on our website. Our support team is much more fluent in these kind of questions.

lilliesAndRoses, 2016-06-27:
Alright! Thanks for quick response.

closed 2016-06-27 by lilliesAndRoses

philips77, 2016-06-27:
You may use the Firmware Revision String char in the DIS
https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.device_information.xml 
I would suggest using that.

lilliesAndRoses, 2016-06-27:
Thanks! That was a good pointer.