nordicsemi/IOS-DFU-Library #14: Crash with 1.0.12

larsblumberg, 2016-09-06:
LegacyDFUPeripheral.init, line 58

`self.logger = LoggerHelper(initiator.logger!)`

`initiator.logger` is `nil` and thus crashes firmware update.

larsblumberg, 2016-09-06:
PS: Minor version updates such as from 1.0.8 to 1.0.12 usually don't have breaking API changes, the version jump however renamed `State` to `DFUState`. Not a big issue, just a formal issue that someone might not expect.

mostafaberg, 2016-09-07:
I'll take a look into that issue, it doesn't seem to reproduce on nRF Toolbox which uses this version, so i'll need to further investigate !

mostafaberg, 2016-09-07:
Regarding `State` Vs. `DFUState`, this seems to be a problem with your cache again, we have only deprecated `State` so using it should just show a deprecation warning, but it should work.
we are planning to enforce `DFUState` on the next non-patch update, i would recommend you try to clear your build directory, rebuild and see if the errors go away regarding the use of the `State` enum

larsblumberg, 2016-09-07:
@mostafaberg: Thanks for clarifying regarding `DFUState`, my comment is then obsolete. I will however stick with 1.0.8 for now, due to that crash.

nRF Toolbox probably has set a logger, that's why it's working for you? We didn't attach a logger. Thanks for investigating!

mostafaberg, 2016-09-07:
No worries, unfortunately i'll be off on vacation until Monday 26th Sep. and won't have internet or a proper working computer till then.
I would suggest that you can either stay on 1.0.8 or try to implement a Bogus logger class that does nothing to un-break it, i'll have a fix up and running as soon as I return, sorry about the delays

larsblumberg, 2016-09-08:
Thanks for availability! Enjoy your holidays.

mostafaberg, 2016-10-04:
We have released  version 2 and finally removed the deprecated enums, I have assigned the logger issue and it will be resolved in the next release!

larsblumberg, 2016-10-04:
Thanks for your notice. Release 2.0 requires Swift 3.0, right? We're still using Swift 2.3 but planning to migrate.

mostafaberg, 2016-10-04:
Yes we hopped over 2.3 due to time constraints, let me know if i can be of any help!

larsblumberg, 2016-10-04:
Thanks for your support but it's probably wiser for us to move to Swift 3.0 rather soon.

mostafaberg, 2016-10-04:
I agree, I would recommend Swift 3.0 too that's why we completely bypassed Swift 2.3

mostafaberg, 2016-10-04:
This issue now resolved in v2.1.2

closed 2016-10-04 by mostafaberg