nordicsemi/IOS-DFU-Library #16: Obj-C

Reggian, 2016-09-15:
Common guys.
Objective C version?
Xcode 8 is killing everything with Swift - Objective-C mixed projects...

Or at least make it not a pain to work with (Subclassing of Swift classes in ObjC not available e.g. DFUPeripheralSelector etc.).

mostafaberg, 2016-09-28:
Are you still experiencing this issue ?
can you show us an example on what errors you are receiving ?

mostafaberg, 2016-10-04:
Closing this in a couple of days due to inactivity and no input from OP

Reggian, 2016-10-04:
Hi @mostafaberg, I have not yet had time to check if you have changed something regarding the `DFUPeripheralSelectorDelegate`. If you have made changes can you maybe link the commit?

mostafaberg, 2016-10-04:
Hello again @Reggian , no I haven't since I need more info, can you show me some code snippets on what you're trying to accomplish and what's going wrong ?

Reggian, 2016-10-10:
Hi @mostafaberg, I will try to explain.

*Predisposition:* I have an ObjC project in which I am upgrading the firmware using the Nordic's iOS DFU Library, which is written in Swift.

*Requirements:* I need custom peripheral selection (i. e. the usage of `DFUPeripheralSelectorDelegate` protocol).

*Issue:* `DFUPeripheralSelectorDelegate` is not marked with `@objc` flag and thus is not exported to the Swift Bridging header. But as I dug deeper I saw that you have replaced all `peripheralSelector` properties from a class implementing the `DFUPeripheralSelectorDelegate` protocol to the concrete `DFUPeripheralSelector` class which can not be subclassed in ObjC code.

*Solution:* Revert all `peripheralSelector` properties back to `id<DFUPeripheralSelectorDelegate>` and expose the `DFUPeripheralSelectorDelegate` protocol (by adding the `@objc` flag)

*PS:* I might end up solving this in my fork and maybe it will be clearer then.

mostafaberg, 2016-10-10:
I see your point, I will take a look and see how fast we can get something like that in place and will update you, problem is we have a lot on our plate right now so this fix might not come out very soon as we would like, but we will probably have it in the near future to support Obj-C projects like yours !, sorry for the inconvenience / delays

mostafaberg, 2016-10-31:
@Reggian I'm getting a new update on the way now, I'll be addin `@objc` to the `DFUPeripheralSelectordelegate` protocol, regarding reverting `peripheralSelector` to `id<DFUPeripheralSelectorDelegate>`, that's not possible in Swift, ObjC only has `id`

`DFUPeripheralSelector` is already an open class so it's subclassable, even though IMO i don't see a reason you should need to subclass it ?

mostafaberg, 2016-10-31:
Fixed, changes will be available from version 2.1.5 and above !

Reggian, 2016-10-31:
> regarding reverting peripheralSelector to id<DFUPeripheralSelectorDelegate>, that's not possible in Swift, ObjC only has id.
Yes, I know. In Swift you do just `DFUPeripheralSelectorDelegate`.

> DFUPeripheralSelector is already an open class so it's subclassable, even though IMO i don't see a reason you should need to subclass it ?
You can not subclass Swift classes in ObjC. Why subclass it? Well, is there another way of making my own filter for which peripherals should get selected?

mostafaberg, 2016-10-31:
What specific things do you need to implement in your filter ?

Reggian, 2016-10-31:
My peripheral advertises specific service data containing an identifier.

mostafaberg, 2016-10-31:
aha, how do you scan for it the first time before starting the DFU process ?

Reggian, 2016-10-31:
I do not know if I understand the question correctly.

I scan it normally by service UUID. The peripheral advertises it. I connect to it and when I want to make a full upgrade (SD, BL and APP) I read a specific identifier so that when the peripheral goes into +1 mac address mode in BL I know that I am connecting to the same peripheral and not some other one that is potentially exposing the DFU protocol.

mostafaberg, 2016-10-31:
Ofcourse, now I understand your issue :)
So when the device restarts you're basically advertising a specific identifier to ensure that it's the correct peripheral.

I'll take a look with @philips77 and see if we can think of any other solution to this and will let you know soon !

Reggian, 2016-10-31:
Thanks @mostafaberg.

philips77, 2016-11-19:
Yes, you're right. It should be a delegate. I'll change it asap.

philips77, 2016-11-19:
Hi, I've changed the selectorPeripheral type to DFUPeripheralSelectorDelegate on branch Version3. Please, use this branch for now. It will be merged into master when we get some feedback.
Now you don't have to override https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUServiceInitiator.swift#L171 method to set the selector, but you can just set it in the initiator and you'll get secure/legacy as a parameter in filter(hint:) method. I hope it can be easier.

closed 2016-12-15 by philips77