nordicsemi/IOS-DFU-Library #17: When will  xcode 8 compitable version  be release?

futomtom, 2016-09-15:


liuyan3176, 2016-09-18:
After update my Xcode from 7.3.1 to 8.0 
the error is in the following:
```
Zip will not be code signed because its settings don't specify a development team.
“Use Legacy Swift Language Version” (SWIFT_VERSION) is required to be configured correctly for targets which use Swift. Use the [Edit > Convert > To Current Swift Syntax…] menu to choose a Swift version or use the Build Settings editor to configure the build setting directly.
```

But when I change the Pods > TARGETS > Zip > Use Legacy Swift Language Version`section from "Unspecified" to "No".There will occur 46 errors...

timrichardsn, 2016-09-19:
@liuyan3176 Until the DFU library and it's Zip dependency are fully updated to Swift 3 you'll have to "Use Legacy Swift Language Version", i.e. Swift 2.3, and fix any errors manually. This is what I've done for now.

liuyan3176, 2016-09-20:
@timrichardsn 
The issue is whether I set the "Use Legacy Swift Language Version" section to "Unspecified" or "Yes" or "No" , the error is still here....

klemenzagar91, 2016-09-20:
I have the same issue regardless of the the selection in "Use Legacy Swift Language Version".

timrichardsn, 2016-09-20:
That's weird. I've set "yes" on use legacy Swift on my project target and all dependencies. Then try:
- Clean project
- Clean build folder
Cleaning the project, and cleaning the  and doing a build DOES give me errors, but they are just syntax conversion errors and only took about 15 minutes or so to fix.

timrichardsn, 2016-09-20:
@liuyan3176 @klemenzagar91 Have you guys tried adding this to the bottom of your Pod file?

`post_install do |installer|
    installer.pods_project.targets.each do |target|
        target.build_configurations.each do |config|
            config.build_settings['SWIFT_VERSION'] = '2.3'
            config.build_settings['MACOSX_DEPLOYMENT_TARGET'] = '10.10'
        end
    end
end`

timrichardsn, 2016-09-20:
@liuyan3176 @klemenzagar91 Have you guys tried adding this to the bottom of your Pod file?

```post_install do |installer|
    installer.pods_project.targets.each do |target|
target.build_configurations.each do |config|
config.build_settings['SWIFT_VERSION'] = '2.3'
config.build_settings['MACOSX_DEPLOYMENT_TARGET'] = '10.10'
end
end
end```

klemenzagar91, 2016-09-20:
I'll try as soon as possible. Thanks for helping

timrichardsn, 2016-09-20:
@liuyan3176 @klemenzagar91 Have you guys tried adding this to the bottom of your Pod file?
```
post_install do |installer|
    installer.pods_project.targets.each do |target|
        target.build_configurations.each do |config|
            config.build_settings['SWIFT_VERSION'] = '2.3'
            config.build_settings['MACOSX_DEPLOYMENT_TARGET'] = '10.10'
        end
    end
end
```

klemenzagar91, 2016-09-20:
I tried it on main project and it didnt work. But there were probably some other errors, that can be caused by other frameworks. I'll try it in a clean project.

liuyan3176, 2016-09-21:
@timrichardsn  thanks for your reply,but these code doesn't work...
I also pod update the Zip to 0.6.0,remove these code or nor, but  also doesn't work,and regardless of the the selection in "Use Legacy Swift Language Version"

liuyan3176, 2016-09-21:
After I update the Zip version to 0.6.0,my project in Xcode 7.3.1 can not work...

philips77, 2016-09-21:
Hi, @mostafaberg who maintaines the library comes back on Monday. Before he left for vacations he was working on removing EV Reflections dependency and migrating the code to Swift 3. The next version should be compatible with xcode 8. Sorry for the delay.

klemenzagar91, 2016-09-21:
what is the temporary solution. I cant use the current code, because I doesnt complie. What should I do, so I can continue my work?

xiejiajian, 2016-09-21:
Hi,what's the relationship between this project and "https://github.com/NordicSemiconductor/IOS-DFU-Library" ? Looks like the latter supports swift 3.
Wait for your new release ! Thanks.

philips77, 2016-09-21:
They should be exactly the same, despite some other folder structure required by Pods. But the development is done on the other project, so perhaps it may have some newer code. Also, there is a pull request with Swift 3 afaik.

mostafaberg, 2016-09-26:
I'm currently working on Swift 3.0 migration, Swift 2.3 will not be supported, the release is targeted to be within this week !, will update this issue with info along the way!
thanks for your patience

mostafaberg, 2016-09-28:
[Swift 3.0 is now supported !](https://cocoapods.org/pods/iOSDFULibrary)

closed 2016-09-28 by mostafaberg