nordicsemi/IOS-DFU-Library #18: i pod update and then convert swift3.0 ,there is error, the pod is not support swift3.0 ?

xRuby, 2016-09-19:


sanderdb01, 2016-09-19:
I am in a similar boat. I have several dozen errors about the update. Will there be updated code? 

FYI, I ran pod update to get the latest code, is that the correct way to do it as of now? Thank you

philips77, 2016-09-19:
Hi, @mostafaberg is coming back on Monday and will handle all issues then.

mostafaberg, 2016-09-23:
The Swift 3.0 PR merge was done erroneously and has been reverted, work will be resumed this coming week and the update will be released in the same week!, at the current time, Swift 3.0 and the latest version of XCode are not supported!

mostafaberg, 2016-09-26:
Work on Swift 3.0 migration is now resumed, will be ready within this week!

mostafaberg, 2016-09-28:
[Swift 3.0 is now supported !](https://cocoapods.org/pods/iOSDFULibrary)

closed 2016-09-28 by mostafaberg

sanderdb01, 2016-09-28:
So to update, should we just do pod update in the terminal after we update our podFile? When I did that, XCode gives me this error:
/Users/david/Projects/iOS Projects/Reverie/Reverie/DFUUpdateView.swift:11:8: Module file was created by an older version of the compiler; rebuild 'iOSDFULibrary' and try again: /Users/david/Library/Developer/Xcode/DerivedData/Reverie-dzbmxgfwzcovxtcabkyfiaygzabv/Build/Products/Debug-iphonesimulator/iOSDFULibrary.framework/Modules/iOSDFULibrary.swiftmodule/x86_64.swiftmodule

Any thoughts on what I might be doing wrong? Thanks a bunch

mostafaberg, 2016-09-28:
@sanderdb01 Yes you'll just need that, but currently there's [an issue](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/20) with v2.0 missing some filenames in the podspec, so it won't work, I'm currently working to get that updated ASAP

sanderdb01, 2016-09-28:
I am also finding that my code can no longer find iOSLibrary when I import it. I look inside the Pods/iOSDFULibrary folder and there is only a README file and a LICENCE file, but no classes folder or anything else. I checked my old commits and that folder was there as well as in your source folder on github. I was wondering if there was something wrong with your CocoaPods setup, or perhaps I did something wrong. Thanks a bunch.

klemenzagar91, 2016-09-28:
In other thread I get a reply. It their mistake and they'll solve it ASAP.

mostafaberg, 2016-09-28:
Yes you'll see the issue in the comment above yours , currently downloading Xcode 8 again to do it from home, hang in there fix is on the way soon :)

mostafaberg, 2016-09-28:
@sanderdb01 Can you give it a try with 2.0.1 now and let me know if things look ok ?

sanderdb01, 2016-09-28:
It loads up the code great this time, thanks a bunch. I am still getting quite a few errors, but so far they seem to be on my end. I am also getting a warning on your files saying that it i smissing a working copy of several of your .swift files, but that might also be on my end. Let me do some cleaning and I will let you know. Thanks again for the great turnaround.