nordicsemi/IOS-DFU-Library #20: When updating to version 2.0 the library disappear.

klemenzagar91, 2016-09-28:
[image]
[image]

I'm also using xcode 8 for the first time. I'll try to solve this by myself, but just so you know there is maybe an issue on your side. As you can see, there is just LICENSE and README file in the folder.

mostafaberg, 2016-09-28:
Ouch, somehow managed to delete the sources in the podspec, will get that fixed ASAP!

mostafaberg, 2016-09-28:
@klemenzagar91  Sorry for the delays, lots of configs, can you verify that it works now ? `2.0.1` wasjust released to fix this issue

klemenzagar91, 2016-09-29:
aaaa it works :)

thank you

mostafaberg, 2016-09-29:
That's great to hear!, thanks a lot for your feedback and sorry for the delays!

closed 2016-09-29 by mostafaberg