nordicsemi/IOS-DFU-Library #21: Zip will not be code signed because its settings don't specify a development team.

xRuby, 2016-09-29:
HI, thank you so quickly update code, and  there is an error  "Zip will not be code signed because its settings don't specify a development team."   how to  resolved it.

pod  log:
Installing Zip (0.6.0)
Installing iOSDFULibrary (2.0.1)

mostafaberg, 2016-09-29:
No worries, This is an Xcode8 feature.

First of all make sure you have the latest cocoapods version installed `sudo gem install cocoapods --pre` and then rerun `pod install`.

If that doesn't work then try the following:
1- Remove our `iOSDFULibrary` from your Podfile
2- Run `pod install` to uninstall it
3- Add it again to your Podfile
4- Run `pod install` another time to add it again
5- Open up Xcode and Clean your build directory by pressing `Option, Shift, Command, K`
6- Build the project and you should be good to go.

if that doesn't work either let us know.

xRuby, 2016-09-29:
Thank you very much,  i resolved it by your method,   thank you answer quickly very much.

mostafaberg, 2016-09-29:
No worries:) have a great day

closed 2016-09-29 by mostafaberg

LtkDev, 2016-10-05:
@mostafaberg 
It is still build failed. I had do that 6 steps as you provide. Something I lost ?
  Xcode:7.3.1 and 
  Installing Zip (0.6.0)
  Installing iOSDFULibrary (2.1.2)

build fail in QuickZip.swift, Zip.swift, ZipUtilities.swift

mostafaberg, 2016-10-05:
Can you open a new issue since it's not the asme issue

xRuby, 2016-10-05:
@LtkDev why not use Xcode8 and swift 3.0 to adaptation IOS 10，i use Xcode 8 and swift 3.0 resolved it by the six steps.