nordicsemi/IOS-DFU-Library #23: Connected peripherals don't work

kklobe, 2016-10-03:
I am working with a buttonless DFU configuration, so I only start the DFU process after I've connected, but the library starts out by trying to connect to the peripheral and never progresses.

It would be nice to have connected peripheral support added. For example, I've added a quick hack to my local copy of DFUExecutor's start function:

```
func start() {
        DispatchQueue.main.async(execute: {
            self.delegate?.didStateChangedTo(DFUState.connecting)
        })
        
        let centralManager = self.initiator.centralManager
        centralManager.delegate = self
        if peripheral.state == .connected {
            peripheral.delegate = self
            // Caller is responsible for having discovered the DFU service and characteristics
            for aService in peripheral.services! {
                initiator.logger?.logWith(.verbose, message: "Discovered Service \(aService.uuid) on peripheral \(peripheral)")
                if aService.uuid == SecureDFUService.UUID {
                    //First priority if SDFU
                    self.didDiscoverDFUService(true)
                    return
                }
                
                if aService.uuid == LegacyDFUService.UUID {
                    //Second priority is legacy DFU
                    self.didDiscoverDFUService(false)
                    return
                }
            }
            //No DFU found at this point, disconnect and report
            self.deviceNotSupported()

        } else {
            centralManager.connect(peripheral, options: nil)
        }
    }

```

mostafaberg, 2016-10-04:
@kklobe That's a good idea, thanks for reporting, we'll get this done in the next release!

mostafaberg, 2016-10-31:
Fix is on it's way now, available from version 2.1.6 and above !

mostafaberg, 2016-11-03:
Reopened after a issue #45 shows that this still doesn't work , probably was due to the fact that it fails due to services were already discovered and not the fact that peripheral was already connected.

kklobe, 2016-11-05:
I think it's a combination - this issue was originally because the connect call never succeeded (because it was already connected), in the same way #45 was failing to discover because the services were already discovered. I've modified DFUExecutor.swift to work with both connected and disconnected states - I'm happy to submit a pull request if you are interested!

philips77, 2016-11-18:
Hi, the Version 3 is now published on our Version3 branch. You can edit your Profile and give the whole repo and branch path there. Could you try with this version?
Version 3 is now in beta phase and we will publish it when we get some feedback.
Thanks!

philips77, 2016-12-15:
I'm closing the issue. Please, feel free to reopen the issue if you still have this problem.

closed 2016-12-15 by philips77