nordicsemi/IOS-DFU-Library #24: Swift 2.3 Support

thedillonb, 2016-10-04:
Consider pulling the following (https://github.com/thedillonb/IOS-Pods-DFU-Library/tree/swift2.3) into a swift2.3 branch in this repository. I can not create a PR unless you create the swift2.3 branch in this repository - if you accept these changes of course.

mostafaberg, 2016-10-04:
Thanks a lot for the great work, we will review your PR and will update you on that, as we were not planning to fully support Swift 2.3 at the moment due to time constraints, but I'll keep this here for a bit and will let you know the decision after we've evaluated it, again thanks a lot !

ezamagni, 2016-10-04:
Great work @thedillonb!
For what it may count, swift 2.3 support would be very useful to me.

chris118, 2016-10-16:
@thedillonb I pulled the swift2.3 branch, but it seems that can't work on iOS10 or iOS9, I used the xcode8 to compile the source with swift 2.3 option, any help? thank you!

thedillonb, 2016-10-16:
@Chris118 what do you mean can't work? I currently use it on IOS 10 so I'm not sure what kind of issue you are seeing.

mostafaberg, 2016-10-31:
We have decided not to go forward with Swift 2.3, sorry for the inconvenience

closed 2016-10-31 by mostafaberg