nordicsemi/IOS-DFU-Library #25: DFU Error - Writing to characteristic failed

furueili, 2016-10-05:
The DFUServiceDelegate never receives a DFUStateCompleted event with version 2.1.1 of the library. During a flashing update, the delegate receives these state changes: DFUStateConnecting, DFUStateStarting, DFUStateUploading.
Please give me some help thx

mostafaberg, 2016-10-05:
Sorry for the inconvenience, firs things first, let's verify that the issue is actually not from the device since it's reporting an error as I see, can you download `nRF Toolbox` from the app store and try to flash the same file and see if it will fail or not ?

furueili, 2016-10-05:
I use nRF Toolbox run DFU is Success !
I am still using old DFU function in Objective-C,
code as below
[image]
[image]

mostafaberg, 2016-10-05:
Can you try to change the line
`initiator.packetReceiptNotificationParamter = [[[NSUserDefaults standardUserDefaults] valueForKey:@"dfu_number_of_packets"] intValue];` 
to
`initiator.packetReceiptNotificationParamter = 1`
and see if it helps ?

furueili, 2016-10-05:
It is working , but takes long time to upload..
Could u tell me that the variable packetReceiptNotificationParamter means and the correct value is?
thx

mostafaberg, 2016-10-05:
Great news !, So yes it's gonna be slower, but that's the real speed your peripheral can accept data at, before setting PRN value, you are sending packets as fast as the phone can send, the peripheral will then have a hard time taking in all that packets and saving them, causing the issue you experience!

PRN (Packet receipt notifications) is the number of packets the phone can send before waiting for the peripheral to respond with a notification saying that it received them and ready to take in more.

The best value you can use is `12`, to debug I usually recommend trying with `1` or `2` first to make it as slow as possible to verify that it's not a speed issue.

you can experiment with values over `12`, but note that even if it works for you, it might still fail on another device for any reason due to it's own limitations.

`12` is the safest number and it also keeps the DFU process from being extremely slow.

glad to have helped, I also recommend trying the [newest version](https://cocoapods.org/pods/iOSDFULibrary) we released yesterday that has more bugfixes

closed 2016-10-05 by mostafaberg