nordicsemi/IOS-DFU-Library #26: Multiple DFU operations at the same time

Alex1478, 2016-10-06:
Hey!

First of all, thank you for this library; it was really easy to integrate the DFU in our application (I love that).
I was wondering if there is a way (without modifiying the library) to execute two DFU operations at the same time; our company is building small sensors that are mounted on the hands of the user (both hands). The user experience would be much better if we could upgrade the firmware of the two sensors at the same time instead of doing it one after the other. 

For now my workaround to make it work was to comment the lines in the **destroy** and **cleanUp** functions. (Avoiding unwrapping nil object)

    func destroy() {
        //centralManager.delegate = nil
        //peripheral!.delegate = nil
        //peripheral = nil
        //delegate = nil
    }

    fileprivate func cleanUp() {
        //dfuService = nil
    }

Thank you!

Alex

mostafaberg, 2016-10-06:
Was just discussing that possibility yesterday with a colleague, I'm not sure how it will work out of the box if you instantiate two classes at the same time, can you test and see how that works out ? if that doesn't work i'll look into it and see the possibility to add a simultaneous DFU feature and will let you know how it'll go

Alex1478, 2016-10-06:
Thanks for the update! 
Following your question: I launched a DFU process simultaneously to both sensors
- This is not working when I start both DFU at **the same exact time**
- This is working when I start the first DFU (receiving progression callbacks) and then I lauch the second DFU right after (first DFU process already started/initiated)

Let me know if you want more details!

mostafaberg, 2016-10-07:
Thanks for testing that, I'll test it out and see, might be some shared variable that's causing that, will see the possibility of adding multiple peripherals to the DFU process so you won't need to instantiate multiple processes at once, for your use case, what would you like the application facing interface to look like ?
The solution that looks pretty straight forward to me is to scan and select two peripherals then send those tho the `DFUServiceInitiator` instead of sending just one peripheral.

if you have another idea we're open to suggestions of course 👍

Alex1478, 2016-10-08:
Hey!

The way I would see it after a quick though; passing an array of `[CBPeripheral]` and the `CBCentralManager` to the `DFUServiceInitiator`.

A nice to have (but I'm not sure if this is something possible):
1) When the DFU process is started, the `CBCentralManager` properties and states are backed up
2) When the DFU process is completed (success or with errors) the `CBCentralManager` is passed back to the application with the original properties and states (like it was before)

cjlarsen, 2017-07-20:
@Alex1478 @mostafaberg Any update on this?  Looked through the documentation and didn't see anything allowing for multiple updates at the same time.  Thanks!

philips77, 2017-07-20:
Hi, imho it should be possible. Each dfu needs a different CBPeripheral (which is obvious) and CBCentralManager (which is not) objects. The peripherals must have been scanned using the proper manager.
I have never tested it though. 
If you intend to update sd+bl+app, and you finish two+ sd+bl updates at the same time services will start scanning for the new bootloaders, and they can exchange targets. Using the upcoming change and sdk14 it will be better regarding this issue.

cjlarsen, 2017-07-20:
@philips77 Keep "Device disconnected unexpectedly" whenever I try to update multiple nrf51822, but they are all run through the same CBCentralManager in my app.  I thought this library connects them to their own individual CBCentralManagers for the update though.

philips77, 2017-07-20:
Each DFU process assigns it's own delegate to the CBCentralManager, so each concurrent process must have it's own instance. The CBPeripheral must be obtained using this manager, otherwise it won't be able to connect to it. I guess, before starting each DFU, you may create a new CBCentralManager and get the peripheral by its uuid. And in your app you can use only your main manager, that you used for scanning or whatever.

tompi, 2017-12-08:
Hi, just a quick confirmation that this works as you say @philips77 

Have only tried with 3 dfus at the same time, but no problems yet.

philips77, 2017-12-08:
Cool! Thanks for the information! So it's time to close the issue :)

closed 2017-12-08 by philips77