nordicsemi/IOS-DFU-Library #27: DFULibrary not loaded

LtkDev, 2016-10-06:
Environment:
    Xcode:8.0
    objective-c
    iPhone: 5s, iOS 8.1.3
and download dependencies
    Installing Zip (0.6.0)
    Installing iOSDFULibrary (2.1.2)

Xcode build succeeded, but running in iPhone fail. 
I got the output log as following. Please tell me what should I do? 

> dyld: Library not loaded: @rpath/DFULibrary.framework/DFULibrary
  Referenced from: /private/var/mobile/Containers/Bundle/Application/85488289-0B51-434C-ACAF-89C3D590D909/HealthCube.app/HealthCube
  Reason: image not found

>

mostafaberg, 2016-10-06:
Have you had installed the legacy library before installing the Cocoapod version ? you might still have the old library lingering in the project settings somewhere

LtkDev, 2016-10-06:
@mostafaberg 
Thanks a lot for your help. My project can build and run after import library !
My next step, do DFU.

mostafaberg, 2016-10-06:
No worries, glad I could help , if you have any more problems just open an issue and we'll be here to help, good luck ! 👍

closed 2016-10-06 by mostafaberg