nordicsemi/IOS-DFU-Library #28: Fail to connect to peripheral after updating the library

heewoonam, 2016-10-06:
I was using the old library from the other repository (1.0.8), and updated it to the newest release as updagrading to Xcode 8 and iOS10.
However I can't get it to work.
On iOS 10, its stuck at connecting to the device (nothing happens), and on iOS9 the following error occurs:
2016-10-06 18:04:23.266 AppName[773:134716] initiator : Optional(<iOSDFULibrary.DFUServiceInitiator: 0x137dd6bb0>)
2016-10-06 18:04:23.268 AppName[773:134716] FirmwareUpdateView ViewDidDisappear
2016-10-06 18:04:23.270 AppName[773:134716] didStateChangedTo state : 0
2016-10-06 18:04:23.270 AppName[773:134716] call back : didStateChangeTo
2016-10-06 18:04:23.271 AppName[773:134716] *** -[iOSDFULibrary.DFUExecutor centralManager:didConnectPeripheral:]: message sent to deallocated instance 0x13919aac0

This is the backtrace:
* thread #1: tid = 0x20e3c, 0x0000000181356bc8 CoreFoundation`___forwarding___ + 700, queue = 'com.apple.main-thread', stop reason = EXC_BREAKPOINT (code=1, subcode=0x181356bc8)
  * frame #0: 0x0000000181356bc8 CoreFoundation`___forwarding___ + 700
    frame #1: 0x0000000181254d1c CoreFoundation`_CF_forwarding_prep_0 + 92
    frame #2: 0x00000001870c4e68 CoreBluetooth`__29-[CBXpcConnection handleMsg:]_block_invoke + 60
    frame #3: 0x0000000100bf5a7c libdispatch.dylib`_dispatch_call_block_and_release + 24
    frame #4: 0x0000000100bf5a3c libdispatch.dylib`_dispatch_client_callout + 16
    frame #5: 0x0000000100bfb4e4 libdispatch.dylib`_dispatch_main_queue_callback_4CF + 2096
    frame #6: 0x0000000181308dd8 CoreFoundation`__CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
    frame #7: 0x0000000181306c40 CoreFoundation`__CFRunLoopRun + 1628
    frame #8: 0x0000000181230d10 CoreFoundation`CFRunLoopRunSpecific + 384
    frame #9: 0x0000000182b18088 GraphicsServices`GSEventRunModal + 180
    frame #10: 0x0000000186505f70 UIKit`UIApplicationMain + 204
    frame #11: 0x00000001000cda48 AppName`main + 140 at AppDelegate.swift:30
    frame #12: 0x0000000180dce8b8 libdyld.dylib`start + 4

Any suggestion on the fix to this?

mostafaberg, 2016-10-06:
It will be hard to tell what's the issue exactly without seeing some code where the issue occurs.
my guess is that you're losing the `CentralManager` for some reason ?

I think this line is the hint:
`message sent to deallocated instance 0x13919aac0`

Try to see what is stored in that address when you repro and let me know, and some code snippet would help a bit

heewoonam, 2016-10-06:
[image]
This is the track of the address.
There is nothing special in the code.
```
    func setSelectedFirmware() {
        let resourceURL: URL = URL(string: resourceURI)!
        let fileName: String = resourceURL.lastPathComponent
        let resourceData: NSData = NSData.init(contentsOf: resourceURL)!
        
        let filePath: String = NSTemporaryDirectory() + fileName
        try? resourceData.write(to: URL(fileURLWithPath: filePath), options: [.atomic])
        
        let fileURL: URL = URL(fileURLWithPath: filePath)
        
        selectedFirmware = DFUFirmware(urlToZipFile: fileURL)
    }
    
    func setInitiator() {
        initiator = DFUServiceInitiator(centralManager: tactosyManager.centralManager, target: peripheral).withFirmwareFile(selectedFirmware)
        NSLog("initiator : \(initiator)")
        initiator.logger = self
        initiator.delegate = self
        initiator.progressDelegate = self
        
        initiator.start()
    }
    
    func logWith(_ level: LogLevel, message: String) {
        NSLog("logWith level : \(level), message : \(message)")
        
        if message == "Disconnected by the remote device" {
            NSLog("message == Disconnected by the remote device")
            
            tactosyManager.reinitCentralManager()
        }
    }
    
    func didStateChangedTo(_ state: DFUState) {
        NSLog("didStateChangedTo state : \(state.rawValue)")

        updateFirmwareProgressViewDelegate.didStateChangeTo(state)
    }
    
    func onUploadProgress(_ part: Int, totalParts: Int, progress: Int, currentSpeedBytesPerSecond: Double, avgSpeedBytesPerSecond: Double) {
        NSLog("onUploadProgress \(progress)")

        updateFirmwareProgressViewDelegate.onUploadingProgress(progress)
    }
    
    func didErrorOccur(_ error: DFUError, withMessage message: String) {
        NSLog("didErrorOccur message : \(message)")
    }
`

mostafaberg, 2016-10-06:
Can you check if the `CentralManager` object in `tactosyManager` and the `tactosyManager` are not being lost too early in the process ? try printing them out in `onUploadProgress` and `didStatechangedTo` and see if they become `nil` during the process

heewoonam, 2016-10-06:
Well, I don't see it becoming nil,
```
2016-10-06 19:07:52.696 AppName[871:156757] initiator : Optional(<iOSDFULibrary.DFUServiceInitiator: 0x139266620>)
2016-10-06 19:07:52.699 AppName[871:156757] CentralManager start: <CBCentralManager: 0x137e6bbd0>
2016-10-06 19:07:52.700 AppName[871:156757] controller: Optional(<iOSDFULibrary.DFUServiceController: 0x139183870>)
2016-10-06 19:07:52.700 AppName[871:156757] FirmwareUpdateView ViewDidDisappear
2016-10-06 19:07:52.702 AppName[871:156757] didStateChangedTo state : 0
2016-10-06 19:07:52.702 AppName[871:156757] didStateChangedTo Centralmanager: Optional(<CBCentralManager: 0x137e6bbd0>)
2016-10-06 19:07:52.703 AppName[871:156757] call back : didStateChangeTo
2016-10-06 19:07:52.709 AppName[871:156757] *** -[iOSDFULibrary.DFUExecutor centralManager:didConnectPeripheral:]: message sent to deallocated instance 0x139277b70
```

mostafaberg, 2016-10-06:
What happens in the `2016-10-06 19:07:52.700 AppName[871:156757] FirmwareUpdateView ViewDidDisappear` call ? maybe it's being deleted there

heewoonam, 2016-10-06:
It's a view to show if a firmware update exists, and if the user presses update, it will move to FirmwareUpdateProgressView and start updating.
Again, it worked fine on Swift 2.2 and DFULibrary 1.0.8.

mostafaberg, 2016-10-06:
Something must have went wrong in your migration process since the library works fine in our expamles, I would put a breakpoint ant whatever line is calling that `2016-10-06 19:07:52.709 AppName[871:156757] *** -[iOSDFULibrary.DFUExecutor centralManager:didConnectPeripheral:]: message sent to deallocated instance 0x139277b70
` and see what has actually been deallocated.

To make it easier I would suggest enabling zombie objects in your `Schemes->Edit Scheme->Diagnostics` menu

heewoonam, 2016-10-06:
Ok thanks, I'll try and post again if I have trouble with it.

mostafaberg, 2016-10-06:
No worries, just drop a line if you have any issues.

heewoonam, 2016-10-07:
Update on the problem,
it seems that the DFUExecutor is being deallocated, for some reason,
```
2016-10-07 17:07:20.930 TactosyDK[579:84122] <iOSDFULibrary.DFUExecutor: 0x137208740>
2016-10-07 17:07:20.932 TactosyDK[579:84122] FirmwareUpdateView ViewDidDisappear
2016-10-07 17:07:20.934 TactosyDK[579:84122] didStateChangedTo state : 0
2016-10-07 17:07:20.934 TactosyDK[579:84122] call back : didStateChangeTo
2016-10-07 17:07:20.943 TactosyDK[579:84122] *** -[iOSDFULibrary.DFUExecutor centralManager:didConnectPeripheral:]: message sent to deallocated instance 0x137208740
```
Any reason for why this may be happening?

mostafaberg, 2016-10-07:
Maybe there are no longer any references to it, try to hold a reference somewhere that will keep it from deallocating

heewoonam, 2016-10-07:
Solved, I did not hold the reference to ```DFUServiceController``` from ```DFUServiceInitiator.start()``` before since I didn't need it, but that caused the deallocation after the update.
Thank you for your help!

closed 2016-10-07 by heewoonam

mostafaberg, 2016-10-07:
No worries at all , glad it works out !