nordicsemi/IOS-DFU-Library #29: centralManager unable to be used after DFU

qvivr-cris, 2016-10-06:
We are allocating our CBCentralManager as so:

myCentralManager = [[CBCentralManager alloc] initWithDelegate:self queue:nil options:@{CBCentralManagerOptionRestoreIdentifierKey : @"MyBTCentral"}];

We perform a DFU and we attempt to reconnect to our device via the device UUID
eg: 
  NSArray *peripheralArray = [myCentralManager retrievePeripheralsWithIdentifiers:@[nsUUID]];
    if([peripheralArray count] > 0)
    {
         [myCentralManager connectPeripheral: peripheralArray[0] options:nil];        
    }
 
At this point, centralManager centralManagerDidUpdateState delegate is called and the centralManager state is  CBManagerStateUnsupported 

Is there a way for us to reset the  centralManager ?

qvivr-cris, 2016-10-06:
(my apologies if this is posted in incorrect forum or location)

mostafaberg, 2016-10-07:
The device's UUID might change after being flashed depending on the new firmware, are you sure that the device isn't broadcasting a different UUID at that time ?

also, are you using `Obj-c` ?

qvivr-cris, 2016-10-07:
Hi, thanks for the response. 
Yes I'm using objective C. 
I should add that killing the App, and relaunching it, in which it re-initialize myCentralManager  it does connect to the device fine.   
I've tried to reinit the myCentralManager without relaunching and again I get CBManagerStateUnsupported.   I could try it without the CBCentralManagerOptionRestoreIdentifierKey to see if that is a factor, but it only would be a test, as we generally do need the use of the CBCentralManagerOptionRestoreIdentifierKey.

mostafaberg, 2016-10-07:
Yeah that was my next tip, can you try to avoid using the options and recreate a new CentralManager and see if it works ? not the solution, just testing to troubleshoot :)

also, can you verify the peripheral has the same UUID you're trying to connect to ? you can use our [nRFConnect](https://itunes.apple.com/no/app/nrf-connect/id1054362403?mt=8) app on the app store to do that

qvivr-cris, 2016-10-07:
Yes, its the same UUID, as we use the UUID when we relaunch the App to connect. 
I will try reinit the myCentralManager without the  CBCentralManagerOptionRestoreIdentifierKey and see what results I get.

mostafaberg, 2016-10-07:
Ok so UUID is not your issue for sure, try without the restoration setting and see if it works out

xiejiajian, 2016-10-08:
I am looking forward to your solution,cause I have the familiar problem, even if i don't use  CBCentralManagerOptionRestoreIdentifierKey.
And i have an additional issue, when i use obj-c to create a 'dfuViewController' just like your dfu Sample, it seems not working as i tapped the 'resume'
Hope there is a guide ,thanks!

`        __weak __typeof__(self) weakSelf = self;
        UIAlertController *alertController = [UIAlertController alertControllerWithTitle:@"Warning" message:@"Are you sure you want to stop the process?" preferredStyle:UIAlertControllerStyleAlert];
        UIAlertAction *cancelAction = [UIAlertAction actionWithTitle:@"NO" style:UIAlertActionStyleCancel handler:^(UIAlertAction * _Nonnull action) {
            [weakSelf logWith:LogLevelVerbose message:@"Abort cancelled"];
            if (weakSelf.dfuController.paused == YES) {
                [weakSelf.dfuController resume];
            }
        }];
        
        UIAlertAction *okAction = [UIAlertAction actionWithTitle:@"YES" style:UIAlertActionStyleDefault handler:^(UIAlertAction * _Nonnull action) {
            [weakSelf logWith:LogLevelVerbose message:@"Abort Confirmed"];
            [weakSelf.dfuController abort];
        }];
`

mostafaberg, 2016-10-10:
@qvivr-cris Do you have any updates yet ?
@xiejiajian please open a new issue to avoid confusion and i'll help you there!

calvingit, 2016-10-11:
After DFU, centralManager's delegate is taken over by DFU Library. So, the original  centralManager can't work with CBCentralManagerDelegate methods. You should reset the delegate of centralManager to belong  to your class.

qvivr-cris, 2016-10-11:
I believe the issue is that we were using CBCentralManagerOptionRestoreIdentifierKey, and initializing a 2nd time, which would fail. A 2nd initialization without a key worked fine, and we were able to reconnect via the device via iOS API of 

NSArray *peripheralArray = [myCentralManager retrievePeripheralsWithIdentifiers:@[nsUUID]];
[myCentralManager connectPeripheral: peripheralArray[0] options:nil];

casey-pax, 2016-10-11:
To @calvingit 's comment, we re-establish the central manager delegate to our own class when the DFU process is done or interrupted. The following code seems to work in all scenarios except one edge-case interruption. This basically listens to when the Nordic SDK is done with the delegate by setting it to nil:

`- (void)startUpdate {
    [myCentralManager addObserver:self forKeyPath:@"delegate" options:NSKeyValueObservingOptionOld | NSKeyValueObservingOptionNew context:nil];

    // start DFU.
}

- (void)observeValueForKeyPath:(NSString *)keyPath ofObject:(id)object change:(NSDictionary<NSString *,id> *)change context:(void *)context
{
    if(object != myCentralManager)
    {
        return;
    }
    
    id newDelegate = change[NSKeyValueChangeNewKey];
    id oldDelegate = change[NSKeyValueChangeOldKey];

    if((newDelegate == nil || [newDelegate isKindOfClass:[NSNull class]]) && oldDelegate != self)
    {
        myCentralManager.delegate = self;
    }
}
`

qvivr-cris, 2016-10-11:
We needed to use CBCentralManagerOptionRestoreIdentifierKey, so we went back to it, just now  identifier we use is now unique each init 

static int identifierIndex = 0;
NSString *restoreIdentifier = [NSString stringWithFormat:@"SWYPCardCentral%i", identifierIndex++];
myCentralManager = [[CBCentralManager alloc] initWithDelegate:self queue:nil options:@{CBCentralManagerOptionRestoreIdentifierKey : restoreIdentifier }];

and we are able to successfully reconnect after a DFU.

calvingit, 2016-10-11:
@qvivr-cris very good.  My situation is a little different with yours. I have wrapped CBCentralManager with a singleton, and wrapped CBPeripheral as another class named [LGPeripheral](https://github.com/LGBluetooth/LGBluetooth).

mostafaberg, 2016-10-11:
I see the issue, using a different `CBCentralManager` to reconnect doesn't work all the time, i guess the right thing to do is to restore the `CBCentralManager` from our side after we've "hijacked" it from the developer, this is an enhancement we had planned and should be out in an upcoming version, glad to hear it worked out

mostafaberg, 2016-10-11:
Possible duplicate https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/6

qvivr-cris, 2016-10-11:
It worked once, on one OS (iOS 10.0.2) we will want to test it more.  Yes is there was a process to hand off ownership and take ownership back that would probably be better.  Thank you.

mostafaberg, 2016-10-12:
Will work on that on the next release and will try to update and will do some tests to verify this works as it should

rushstrike, 2017-03-03:
any work around on restoring the delegate after the DFU class hijack it?

mostafaberg, 2017-03-03:
@rushstrike There's a good example solution in the comments above [found here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/29#issuecomment-252816614)

philips77, 2018-10-05:
This will be invalid after #228 gets merged.

philips77, 2018-10-05:
Fixed in https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/releases/tag/4.2.0

closed 2018-10-05 by philips77