nordicsemi/IOS-DFU-Library #30: Peripheral disconnects after discovering characteristics for DFUService

JohnGroenhof, 2016-10-10:
Hello, 
I just updated to the Swift 3 DFU library, and I am having an issue where my peripheral gets disconnected every time when the DFU Service tries to discover characteristics for the DFUService. There are a few details that seems specific to this issue: 

- The app is written in Objective-C.
- The code I'm using worked prior to updating the DFU Library, and prior to iOS10. 
- This same code works on an iOS9 device, so this disconnection is ONLY happening on iOS10. 
- I've tested my peripheral and my firmware ZIP with the DFU-Library example app on iOS10, and that works as well. So I'm only seeing this issue on iOS10 when the DFU-library is called from Objective-C code. 
- I'm using the Legacy DFU service. 

The process I'm taking is: connect to device, discover Legacy DFU service, write 0x0104 to reboot into bootloader. Connect to DFUTarg, then start the DFU process via the following function:

```
- (void)startDFU {    
    if ( [BTManager sharedInstance].connectedPeripheral != nil && self.firmwareZIPForDFU != nil ) {
        DFUServiceInitiator *dfuInitiator = [[DFUServiceInitiator alloc] initWithCentralManager:[BTManager sharedInstance].centralManager target:[BTManager sharedInstance].connectedPeripheral];
        
        [dfuInitiator withFirmwareFile: self.firmwareZIPForDFU];
        [dfuInitiator setLogger: self];
        [dfuInitiator setDelegate: self];
        [dfuInitiator setProgressDelegate: self];
        self.dfuController = [dfuInitiator start];
    }
}
```

The DFU Service seems to start normally, but every time I run it, the peripheral disconnects at the exact same place. Here is the log starting from my connection to DFUTarg:

> 2016-10-10 13:46:39.497344 DFUTest[894:338968] Connected peripheral with name: DfuTarg
> 2016-10-10 13:46:39.497418 DFUTest[894:338968] -[ViewController didConnectDFUTarg]
> 2016-10-10 13:46:39.497469 DFUTest[894:338968] -[ViewController startDFU]
> 2016-10-10 13:46:39.501960 DFUTest[894:338728] DFU Service didStateChangedTo: 0
> 2016-10-10 13:46:39.505383 DFUTest[894:338973] LoggerDelegate logWith: 1 message: Discovering all services for peripheral <CBPeripheral: 0x1740fd500, identifier = 5CB14D36-3993-412F-A3FD-B15E138F088E, name = DfuTarg, state = connected>
> 2016-10-10 13:46:39.879373 DFUTest[894:338968] LoggerDelegate logWith: 1 message: Discovered Service 00001530-1212-EFDE-1523-785FEABCD123 on peripheral <CBPeripheral: 0x1740fd500, identifier = 5CB14D36-3993-412F-A3FD-B15E138F088E, name = DfuTarg, state = connected>
> 2016-10-10 13:46:39.883675 DFUTest[894:338968] LoggerDelegate logWith: 1 message: Did discover legacy DFU service
> 2016-10-10 13:46:39.883864 DFUTest[894:338968] LoggerDelegate logWith: 1 message: Starting legacy DFU Service initiator
> 2016-10-10 13:46:39.885382 DFUTest[894:338968] LoggerDelegate logWith: 1 message: Instantiated Legacy DFU peripheral selector
> 2016-10-10 13:46:39.887011 DFUTest[894:338968] LoggerDelegate logWith: 1 message: Connecting to DfuTarg...
> 2016-10-10 13:46:39.887126 DFUTest[894:338728] DFU Service didStateChangedTo: 0
> 2016-10-10 13:46:39.887507 DFUTest[894:338968] LoggerDelegate logWith: 0 message: centralManager.connectPeripheral(peripheral, options:nil)
> 2016-10-10 13:46:39.888250 DFUTest[894:338968] LoggerDelegate logWith: 1 message: Started Legacy DFU service controller
> 2016-10-10 13:46:39.891766 DFUTest[894:338973] LoggerDelegate logWith: 0 message: [Callback] Central Manager did connect peripheral
> 2016-10-10 13:46:39.891974 DFUTest[894:338973] LoggerDelegate logWith: 5 message: Connected to DfuTarg
> 2016-10-10 13:46:39.893759 DFUTest[894:338973] LoggerDelegate logWith: 1 message: Discovering services...
> 2016-10-10 13:46:39.894073 DFUTest[894:338973] LoggerDelegate logWith: 0 message: peripheral.discoverServices(nil)
> 2016-10-10 13:46:39.897030 DFUTest[894:338968] LoggerDelegate logWith: 5 message: Services discovered
> 2016-10-10 13:46:39.897466 DFUTest[894:338968] LoggerDelegate logWith: 1 message: DFU Service found
> 2016-10-10 13:46:39.897770 DFUTest[894:338968] LoggerDelegate logWith: 1 message: Discovering characteristics in DFU Service...
> 2016-10-10 13:46:39.897893 DFUTest[894:338968] LoggerDelegate logWith: 0 message: peripheral.discoverCharacteristics(nil, forService:DFUService)
> 2016-10-10 13:46:40.115262 DFUTest[894:338973] LoggerDelegate logWith: 0 message: [Callback] Central Manager did disconnect peripheral
> 2016-10-10 13:46:40.115430 DFUTest[894:338973] LoggerDelegate logWith: 5 message: Disconnected
> 2016-10-10 13:46:40.120561 DFUTest[894:338973] LoggerDelegate logWith: 20 message: Error 6: The connection has timed out unexpectedly.
> 2016-10-10 13:46:40.120947 DFUTest[894:338728] DFU Service didErrorOccur: 202 withMessage: The connection has timed out unexpectedly. (code: 6)

mostafaberg, 2016-10-10:
can you try setting the packet receipt notification parameter to 12 and see if that helps ?

JohnGroenhof, 2016-10-10:
@mostafaberg , after setting the packetReceiptNotificationParameter to 12, I was able to get 2 successful DFU's, but about 95% of the time, I was seeing the same error. I tried changing this value to many different numbers, but still get a disconnect from the device at the same place.

mostafaberg, 2016-10-11:
Interesting, how about setting it to 1 and see if you manage to reproduce, with prn set to 1 the connection will be very slow, but this is just to debug

JohnGroenhof, 2016-10-11:
Setting packetReceiptNotification to 1 did not change my results, however I think I've found a working solution. It appears that CoreBluetooth is running a little bit slow for me on iOS10. When I wrote 0x0104 to the legacy DFU service to reboot the peripheral into bootloader, I would immediately start scanning for peripherals with the legacy DFU UUID, and as soon as it connected to DFUTarg, call the startDFU function. Now, if I use a dispatch_after call to delay the startDFU function by a second or two, DFU works. 

It is very odd that I'm only having this issue on iOS10. Have you guys seen any performance changes with CoreBluetooth or CoreLocation in iOS10? Feel free to close this issue, I am able to perform DFU now with this little hack in place.

mostafaberg, 2016-10-31:
Yes there are some bugs here and there in iOS 10, but most have been handled in the 10.1.x betas, can you update to the latest iOS 10.1 and see if the issue still reproduces ?

mostafaberg, 2017-02-07:
Closing this as no longer relevant, please reopen a new issue if problem still persists

closed 2017-02-07 by mostafaberg