nordicsemi/IOS-DFU-Library #32: 'Zip' dependency not handled for Carthage users

jakerockland, 2016-10-12:
Currently this library doesn't have a Cartfile to address the 'Zip' dependency for Carthage users. I've made a small patch at #31.

jakerockland, 2016-10-12:
This is handled I believe by #31 if the Examples directory is deleted. Otherwise, there will be a build error due to a conflict between all of the files from the 'Zip' library that have been manually copied to the Examples directory and the library itself when checked out with Carthage. Why is the Examples directory something included in the library?

mostafaberg, 2016-10-18:
Yes this seems to be the case, will review your PR and update the fix hopefully during this week!

jakerockland, 2016-10-18:
Awesome, thanks so much @mostafaberg 👍 😄

mostafaberg, 2016-10-31:
Fixed now with help from @jakerockland 's PR #33 :)

closed 2016-10-31 by mostafaberg