nordicsemi/IOS-DFU-Library #34: DFUServiceController resume not work

LtkDev, 2016-10-14:
Hi, I had use [controller resume] , but it's not update firmware continued.
Environment: objective-c, iOS 8.1.3, iOSDFULibrary (2.1.2).
I got the output log as following. Please tell me what should I do?
> Progress: 26% (part 1/1). Speed: 65574.422513 bps, Avg speed: 2538.034336 bps
> Progress: 27% (part 1/1). Speed: 56186.255861 bps, Avg speed: 2522.083538 bps
> Progress: 28% (part 1/1). Speed: 66894.800638 bps, Avg speed: 2541.240389 bps
> LogLevel:0, [Callback] Central Manager did disconnect peripheral
> LogLevel:5, Disconnected
> LogLevel:20, Error 6: The connection has timed out unexpectedly.
> DFUServiceDelegate Error -->202: The connection has timed out unexpectedly. (code: 6)

Joyce306, 2016-10-14:
May be the signal of Bluetooth interference, there are a number of Bluetooth devices around me, including mobile phones and so on, my device sometimes disconnected

LtkDev, 2016-10-15:
@Joyce306 
After [controller resume], it was wait few time and no response util timed out.
I had download library by Carthage.

mostafaberg, 2016-10-18:
@LtkDev have you tried setting the PRN value to 12 ?

LtkDev, 2016-10-21:
@mostafaberg 
Yes, I did it.
initiator.packetReceiptNotificationParameter = 12;

mostafaberg, 2016-10-21:
Try setting it to a very low value like 1 or 2 and see if that stops those interruptions, this will slow down the flashing process too, just want to see how it goes at a slow rate.

Which Chip/DK are you using ?

LtkDev, 2016-10-24:
@mostafaberg 
SDK v11 & nRF52 s132 v2.0.0

> Progress: 6% (part 1/1). Speed: 331.669754 bps, Avg speed: 326.783546 bps
> Progress: 7% (part 1/1). Speed: 331.554067 bps, Avg speed: 326.122361 bps
> Progress: 8% (part 1/1). Speed: 336.027191 bps, Avg speed: 326.314432 bps
> Progress: 9% (part 1/1). Speed: 329.684831 bps, Avg speed: 325.239866 bps

mostafaberg, 2016-10-24:
it starts at 6% ? can you show the entire log from start till it stops ?

LtkDev, 2016-10-24:
@mostafaberg 

> 2016-10-24 15:59:46.223 LogLevel:10, Initialize DFU Parameters completed
> 2016-10-24 15:59:46.223 LogLevel:1, Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> 2016-10-24 15:59:46.224 LogLevel:0, peripheral.writeValue(0x080100, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
> 2016-10-24 15:59:46.225 Uploading...
> 2016-10-24 15:59:46.282 LogLevel:5, Data written to 00001531-1212-EFDE-1523-785FEABCD123
> 2016-10-24 15:59:46.283 LogLevel:10, Packet Receipt Notif Req (Op Code = 8, Value = 1) request sent
> 2016-10-24 15:59:46.284 LogLevel:1, Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
> 2016-10-24 15:59:46.284 LogLevel:0, peripheral.writeValue(0x03, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
> 2016-10-24 15:59:46.340 LogLevel:5, Data written to 00001531-1212-EFDE-1523-785FEABCD123
> 2016-10-24 15:59:46.341 LogLevel:10, Uploading firmware...
> 2016-10-24 15:59:46.341 LogLevel:1, Sending firmware DFU Packet characteristic...
> 2016-10-24 15:59:48.051 Progress: 1% (part 1/1). Speed: 330.572835 bps, Avg speed: 339.237233 bps
> 2016-10-24 15:59:49.821 Progress: 2% (part 1/1). Speed: 333.288954 bps, Avg speed: 333.411995 bps
> 2016-10-24 15:59:51.591 Progress: 3% (part 1/1). Speed: 332.540811 bps, Avg speed: 331.469045 bps
> 2016-10-24 15:59:53.391 Progress: 4% (part 1/1). Speed: 334.353020 bps, Avg speed: 329.114100 bps
> 2016-10-24 15:59:55.161 Progress: 5% (part 1/1). Speed: 339.351161 bps, Avg speed: 328.826446 bps
> 2016-10-24 15:59:56.960 Progress: 6% (part 1/1). Speed: 333.083499 bps, Avg speed: 327.728147 bps
> 2016-10-24 15:59:58.731 Progress: 7% (part 1/1). Speed: 333.427721 bps, Avg speed: 327.703004 bps
> 2016-10-24 16:00:00.471 Progress: 8% (part 1/1). Speed: 333.945389 bps, Avg speed: 328.394697 bps
> 2016-10-24 16:00:02.241 Progress: 9% (part 1/1). Speed: 333.450252 bps, Avg speed: 328.315640 bps
> 2016-10-24 16:01:03.472 LogLevel:0, [Callback] Central Manager did disconnect peripheral
> 2016-10-24 16:01:03.472 LogLevel:5, Disconnected
> 2016-10-24 16:01:03.474 LogLevel:20, Error 6: The connection has timed out unexpectedly.
> 2016-10-24 16:01:03.475 DFUServiceDelegate Error -->202: The connection has timed out unexpectedly. (code: 6)

mostafaberg, 2016-10-24:
That's interesting indeed, does this work properly when using the nRF Toolbox app ?

LtkDev, 2016-10-24:
@mostafaberg  it was failed.
[image]

mostafaberg, 2016-10-24:
Can you try the same thing on nRF Toolbox for Android ? I just want to confirm if this is an issue related to iOS or not

mostafaberg, 2016-10-24:
@philips77 let me know if you see something I'm missing here 👍

LtkDev, 2016-10-24:
@mostafaberg  In android was successful.
[image]

mostafaberg, 2016-10-24:
Great, can you now try again with iOS and send me the log one final time ? i'll take a look and see what goes wrong there afterwards.

also if you can send me an example project that reproduces the error that would be very helpful:)

LtkDev, 2016-10-24:
@mostafaberg 
In android was successful.
version:1.16.3

mostafaberg, 2016-10-24:
The easiest way to get a fix out is if you create an example code in a public repo and send me a link, i'll reproduce the issue and fix it with a pull request and let you know what's wrong exactly, can you create something small as an example ?

philips77, 2016-12-15:
Hi, we have released version 3.0.1 today. Do you still experience this issue with the new version? Can we close the issue?

philips77, 2017-01-10:
Hi, I'm closing the issue. If you still have problems with it, please reopen or create another issue.

closed 2017-01-10 by philips77