nordicsemi/IOS-DFU-Library #35: Cannot connect with peripheral

sewerynplazuk, 2016-10-16:
Hi,
I am developing application and trying to use this library to implement FOTA into it. I am using Swift 3.0 and testing everything on device with iOS 10. Peripheral is based on nRF51822 SoC. 

Unfortunatelly library cannot connect with peripheral - stucks on "Connecting".

I tried running initiator.start() method in different condition (connected with the device, disconnected with the device, in DFU mode, in normal mode) nothing works. Do you have any idea what am I missing?

bluetooth.manager and bluetooth.connectedPeripheral are not nil.

My implementation look   ##like:
```
        let bluetooth = SensorManager.sharedInstance
        self.downloadUpdateFile(uuid: forUUID) { (localFirmwareUrl) in
            if let selectedFirmware = DFUFirmware(urlToZipFile: localFirmwareUrl) {
                let initiator = DFUServiceInitiator(centralManager: bluetooth.manager, target: bluetooth.connectedPeripheral).withFirmwareFile(selectedFirmware)

                initiator.packetReceiptNotificationParameter = 12
                initiator.forceDfu = true
                initiator.logger = self
                initiator.delegate = self
                initiator.progressDelegate = self
                print("START")
                let controller = initiator.start()
            }
        }
```

mostafaberg, 2016-10-18:
This seems to be related to a Bug in iOS10, it has been resolved from Beta 2 and above, can you verify after installing the beta OS ?

sewerynplazuk, 2016-10-18:
Do you mean iOS 10.0 build 2 or 10.1 build 2?

mostafaberg, 2016-10-18:
That's `10.1`, sorry didn't mention the exact version :)

sewerynplazuk, 2016-10-18:
I will verify it as soon as I can and let you know.

sewerynplazuk, 2016-10-19:
I just tested app on device with iOS10.1 beta 4 and it still stucks at "Connecting".

mostafaberg, 2016-10-19:
Sorry that you went through the update with no success, can you send the logger output so I can investigate the issue?

sewerynplazuk, 2016-10-19:
I would but there is no logger output. "Connecting" is the only print I got while running the application.

marcussfu, 2016-10-20:
we have the same issue too, but we skip centralManager.connect(didConnect and didDiscoverServices), then call self.didDiscoverDFUService(false) in DFUExecutor.start().

As a result, we can update the firmware successfully.

However,we don't know whether the method we mentioned above will cause another issue.
@mostafaberg @sewerynplazuk

mostafaberg, 2016-10-20:
@sewerynplazuk  There will always be log events, if you don't see them then you probably haven't setup the logger delegate properly, have you implemented the logger delegate protocol on that class ?

Another thing to try is to disconnect the peripheral before doing that, the library assumes the peripheral is disconnected, it might end up getting stuck on connecting, try to disconnect first and see, we have a fix for that on the way that'll be in the next release.

@marcussfu it is a workaround that involves modifying the source code of the library, which makes it hard for us to support, I would try to do the same thing too, disconnect first and then initiate the DFU process and see if it works out

sewerynplazuk, 2016-10-20:
@marcussfu I'll try it, thanks.
@mostafaberg I will check logger once more and give you feedback but you have to wait until Monday.

I tried running start() method while peripheral was disconnected - it's not the case.

marcussfu, 2016-10-21:
@sewerynplazuk 
Forgot to mention, we first connect bluetooth with the device before DFUExecutor.start(),  then skip centralManager.connect, and call self.didDiscoverDFUService(false).

mostafaberg, 2016-10-21:
Ok keep me updated !

sewerynplazuk, 2016-10-24:
@mostafaberg, I double checked everything and logger delegate is properly set although it just don't print any logs. Method logWith() hasn't fired even once during my test. 

Workaround provided by @marcussfu works great - thanks once again. I am able to perform update now (logs are working too). I think that you should consider to implement this approach into next release.

mostafaberg, 2016-10-24:
A bit strange to hear about that logging issue, will look into that soon, 
in the next version as i said i'll include the fix as @marcussfu says by making sure things work when the peripheral is already connected.

thanks for your patience :)

DannyWithJoy, 2016-10-26:
@mostafaberg , I faced the same issue today. It took me 2 hours to find the reason.
In my OC code, this is not OK:
`[self.initiator start];`
this is OK:
`self.controller = [self.initiator start];`
I'm guess that:
`DFUServiceController` object return by `-[DFUServiceInitiator start];` has been autoreleased later, so after `conecting...`, `DFUServiceController` object no looger exsit.

It's my case. I hope it will give you some help.

mostafaberg, 2016-10-26:
@DannyWithJoy yes this is the expected behaviour since ObjC Uses ARC so you need to hold a reference to the controller object.

This is a bit irrelevant to that bug though, but thanks a lot for the tip, will surely help someone else

mostafaberg, 2016-10-31:
I've released a new fix that will handle states where peripheral is already connected, can you try that and see if you can still reproduce the bug ? if not i'll need to take another look, let me know
Fix version is `2.1.6`

marcussfu, 2016-11-02:
update to 2.1.6, still reproduce the bug that stucks on "Connecting".

mostafaberg, 2016-11-02:
Ok this doesn't reproduce here for some reason, can you supply an example project that reproduces the issue and i'll make sure to get that done right away

mostafaberg, 2016-11-03:
@marcussfu can confirm i was testing with a completely different mindset on what the issue is, i can confirm this is not fully resolved yet and will try to get the fix out as soon as we can !

philips77, 2016-12-15:
Hi, we have released version 3.0.1 today. Do you still experience this issue, or can it be closed?

sewerynplazuk, 2016-12-16:
I'll check it and let you know.

closed 2017-01-12 by philips77