nordicsemi/IOS-DFU-Library #36: Example for iOSDFULibrary dont works with iOS 10

JoseMiguelAlcaraz, 2016-10-18:
Hi,

I'm testing the example for iOSDFULibrary with my XCode 8 and iOS 10 but always repeat the same loop:

Could you help me?

I : Changed state to: Connecting
V : Discovering all services for peripheral <CBPeripheral: 0x1740e1480, identifier = F48B99D1-CE34-74EE-60DC-446BC5A4723E, name = iMove, state = connected>
V : Discovered Service 6E400001-B5A3-F393-E0A9-E50E24DCCA9E on peripheral <CBPeripheral: 0x1740e1480, identifier = F48B99D1-CE34-74EE-60DC-446BC5A4723E, name = iMove, state = connected>
V : Discovered Service Battery on peripheral <CBPeripheral: 0x1740e1480, identifier = F48B99D1-CE34-74EE-60DC-446BC5A4723E, name = iMove, state = connected>
V : Discovered Service Device Information on peripheral <CBPeripheral: 0x1740e1480, identifier = F48B99D1-CE34-74EE-60DC-446BC5A4723E, name = iMove, state = connected>
V : Discovered Service 00001530-1212-EFDE-1523-785FEABCD123 on peripheral <CBPeripheral: 0x1740e1480, identifier = F48B99D1-CE34-74EE-60DC-446BC5A4723E, name = iMove, state = connected>
V : Did discover legacy DFU service
V : Starting legacy DFU Service initiator
V : Instantiated Legacy DFU peripheral selector
V : Connecting to iMove...
D : centralManager.connectPeripheral(peripheral, options:nil)
V : Started Legacy DFU service controller
I : Changed state to: Connecting
D : [Callback] Central Manager did connect peripheral
I : Connected to iMove
V : Discovering services...
D : peripheral.discoverServices(nil)
I : Services discovered
V : DFU Service found
V : Discovering characteristics in DFU Service...
D : peripheral.discoverCharacteristics(nil, forService:DFUService)
I : DFU characteristics discovered
V : Reading DFU Version number...
D : peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)
I : Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):0100
A : Version number read: 0.1
V : Enabling notifiactions for 00001531-1212-EFDE-1523-785FEABCD123...
D : peripheral.setNotifyValue(true, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123)
I : Changed state to: Starting
V : Notifications enabled for 00001534-1212-EFDE-1523-785FEABCD123
A : DFU Control Point notifications enabled
W : Application with buttonless update found
V : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D : peripheral.writeValue(0x0104, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
I : Changed state to: Enabling DFU Mode
I : Data written to 00001531-1212-EFDE-1523-785FEABCD123
A : Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
D : [Callback] Central Manager did disconnect peripheral
I : Disconnected by the remote device
V : Connecting to iMove...
D : centralManager.connectPeripheral(peripheral, options:nil)

mostafaberg, 2016-10-18:
Can you show a code snippet on how you're initializing the library ?

JoseMiguelAlcaraz, 2016-10-18:
I'm using your example "DFUViewController.swift"

And I lunch the viewcontroller from my Objective-C code 

DFUViewController * dfuVC = [[UIStoryboard storyboardWithName:@"Main" bundle:NULL] instantiateViewControllerWithIdentifier:@"DFUVCID"];
    
    [dfuVC setTargetPeripheralWithAPeripheral:self.btmanager.carPeripheral];
    [dfuVC setCentralManagerWithCentralManager:self.btmanager.centralManager];
    [dfuVC setFirmwareFileNameWithFileName:currentFirmware];
    
    [self presentViewController:dfuVC animated:YES completion:^{
        
    }];

where "currentFirmware" in setFirmwareFileNameWithFileName method is my zip file name.

mostafaberg, 2016-10-31:
Are you sure you're holding a reference to the `DFUExecutor` ?

your code is Obj-C and not Swift so things can be a bit different for you.

try holding a reference to the `DFUExecutor` and see how it goes

mostafaberg, 2016-11-17:
Any updates on that issue ?

philips77, 2017-01-10:
Hi, the lib has been tested on iOS 10 and it works. I'm closing the issue as it hasn't been updated for a while now.

closed 2017-01-10 by philips77