nordicsemi/IOS-DFU-Library #38: current bundled example firmware not compatible with example bootloader

jakerockland, 2016-10-18:
The `.zip` of the secure bundled firmware that is currently used in the example project ([sdfu_test_app_hrm_s132.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/537586/sdfu_test_app_hrm_s132.zip)) does not seem to be compatible with the example bootloader provided by Nordic for the development kit.

When trying to run the example project, I would always get an error that ended with something like:
```
D : peripheral.writeValue(0x04, forCharacteristic: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: WithResponse)
I : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x):600405
A : Response (Op Code = Execute, Status = Invalid object) received
E : Error Optional("Invalid object"): Optional("Invalid object")
E : Error occured: invalidObject, Invalid object
E : Invalid object
E : Invalid object
I : Changed state to: Failed
V : Disconnecting...
D : centralManager.cancelPeripheralConnection(peripheral)
D : [Callback] Central Manager did disconnect peripheral without error
I : Disconnected
A : Operation Aborted by user, peripheral has been disconnected
I : Changed state to: Aborted
```

Updating the bundled firmware to this recent example firmware ([dfu_test_app_hrm_s132.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/537593/dfu_test_app_hrm_s132.zip)) from Nordic seemed to remedy the situation.

It's a pretty simple fix, but I've included the changes we've made on our fork if you want to compare or pull the changes into the main repo: https://github.com/SomaticLabs/IOS-Pods-DFU-Library/pull/2

mostafaberg, 2016-10-31:
Thanks for the tip, updated the firmware for the next version, considering to actually not bundle any firmwares and let the users select their own files to avoid confusion, will see :)

closed 2016-10-31 by mostafaberg

jakerockland, 2016-10-31:
@mostafaberg That might be the best way to go 👍