nordicsemi/IOS-DFU-Library #39: initWithUrlToZipFile return nil

DannyWithJoy, 2016-10-19:
My Installation : 
1. build IOS-Pods-DFU-Library
2. Drag iOSDFULibrary.framework to my objective-C project
3. Set iOSDFULibrary.framework as optional in Linked Frameworks and Libraries

My Enviorments:
Macos 10.12
Xcode 8.0

Now:
I use the demo firmware "dfu_test_app_hrm_s110.zip" to test the function.
But -initWithUrlToZipFile always return nil.
Am I doing something wrong here?

1.Get URL. Return value is correct. (fileURL = file:///var/mobile/Containers/Data/Application/14F7C7B3-A2F2-4704-87B6-1A5E6EC7C2C9/Library/Caches/GOOTA/ota.zip)

`NSURL *fileURL = [NSURL fileURLWithPath:OTA_GO_FIRMWARE_FILEPATH];`

2.Test if the file is empty. Return value is correct, it's not empty.

`NSData *fileData = [NSData dataWithContentsOfURL:fileURL];`

3.Creat DFUFirmware object. Return value is nil.

`DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:fileURL];`

mostafaberg, 2016-10-20:
hi @DannyWithJoy , Can you setup a breakpoint at the [DFUFirmware alloc] initWithUrlToZipFile:] and step into it and show me where the code goes ?
Also which version of this library are you using ?

DannyWithJoy, 2016-10-20:
@mostafaberg ,  sorry, a very tough day for me today!!!

I'm using iOSDFULibrary.framework, so that I can not step into [DFUFirmware alloc] initWithUrlToZipFile:]. Right?

The version of the library I am using: I build the latest code [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library) to get the framework.

Here is a totally clean[ sample code](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library) for you to understanding what I am facing.

Thanks.

mostafaberg, 2016-10-20:
No worries, sorry I missed the part saying that you are using a prebuilt framework, yes you cannot step through that.

We don't officially support building frameworks this way as we noticed it gets a bit problematic for many people so we stopped supporting it as an official method of getting the library.

can you try removing the framework files and install the library using Cocoapods ?

mostafaberg, 2016-10-20:
Also, your sample code link was to the Library's page , can you check it out and give me a correct link ?

DannyWithJoy, 2016-10-20:
My bad, sample code is [here](https://github.com/DannyWithJoy/DFUTest).

DannyWithJoy, 2016-10-20:
About the cocoapods, I tried.
But <user frameworks!> and <Swift support> always bring a lot of building errors in my OC project.
After fews hours, I give it up.
So if you give me a 'OC project with DFUSDK sample code', it will do me a huge favor.

mostafaberg, 2016-10-20:
Ok i'm currently looking into it, my initial guess is that some dep. is missing in the .framework file, will verify and let you know soon

DannyWithJoy, 2016-10-20:
@mostafaberg  Great news. It's a big step forward for us.

mostafaberg, 2016-10-20:
Ok i resolved the problem and sent you  a pull request that you can find [here](https://github.com/DannyWithJoy/DFUTest/pull/1)

Let me know if there's anything I can help with!

DannyWithJoy, 2016-10-20:
@mostafaberg OK.
Checking right now.

DannyWithJoy, 2016-10-20:
@mostafaberg It works. Thanks for your help。 We can finally move on to the next step.
By the way, we strongly recommend that give us a `download address for latest framework file`.

mostafaberg, 2016-10-21:
That's great !, we do not have an address but it's very easy to build them yourself with a single command!, here's how to go it:

1. On your mac please install carthage ([instructions](https://github.com/Carthage/Carthage#installing-carthage))

2. Create a file named `cartfile` anywhere on your computer
3. add the following content to the file:
```
github "NordicSemiconductor/IOS-Pods-DFU-Library" ~> 2.1.2
github "marmelroy/Zip" ~> 0.6
```

4. Open a new terminal and `cd` to the directory where the file is
5. Enter the command `carthage update --platform iOS`
6. Carthage will now take care of building your frameworks, the produced .framework files will be found in a newly created directory called `Carthage/Build/iOS`, copy over `iOSDFULibrary.framework` and `Zip.framework` to your project and you are good to go.

Note: Carthage also produces `iOSDFULibrary.framework.dsym` and `Zip.framework.dsym`, you  can use those files to desymbolicate crashes that are related to the relevant library. but it's not required

closed 2016-10-21 by mostafaberg

zhangferry, 2016-12-14:
I spent so much time to work on this problem, but I figure out with yours help.Thanks you so much😁

pjl1234567, 2017-03-01:
I used your method, install carthage and create a file named cartfile,........ last get  iOSDFULibrary.framework and Zip.framework。    but running failure,

dyld: Library not loaded: @rpath/Zip.framework/Zip
  Referenced from: /var/containers/Bundle/Application/CBF1A722-9957-4E20-8058-287554EC4437/SwiftOC.app/SwiftOC
  Reason: image not found
(lldb)

mostafaberg, 2017-03-01:
@pjl1234567 Please make sure you follow all the steps correctly and that you read the solutions posted here, if that doesn't work please open up a new issue and let us know all the versions you are using (MacOS, XCode, Library, etc..)