nordicsemi/IOS-DFU-Library #42: Cannot abort DFU controller

jakerockland, 2016-10-25:
When trying to abort my DFU controller like this, I'm getting a runtime error:

```
 var controller: DFUServiceController!
 let selectedFirmware = DFUFirmware(urlToZipFile:firmwareURL)
 let initiator = DFUServiceInitiator(centralManager: self.central, target: self.peripheral).withFirmwareFile(selectedFirmware!)
                
initiator.logger = self
initiator.delegate = self
initiator.progressDelegate = self

self.controller = initiator.start()

_ = self.controller.abort()
```

[image]

jakerockland, 2016-10-28:
@mostafaberg Could I get your thoughts on this?

mostafaberg, 2016-10-31:
if you try to abort the controller before it starts it has not discovered at that moment yet if it's a Secure or Legacy DFU so there's nothing to abort per se, however we should handle this more gracefully

does this happen during the even if you abort in the middle of the DFU operation ?

jakerockland, 2016-11-03:
Hmm, that makes sense @mostafaberg, I haven't had this issue when trying to abort mid-DFU but have tried to use the .abort() function when hanging in the "disconnecting" state as noted in #43, which is where I noticed this behaviour.

mostafaberg, 2016-11-03:
True, i can confirm this happens, i'll try to get that resolved as soon as I can, will have much more time soon assigned to fixing all those bugs ! thanks for your patience

jakerockland, 2016-11-03:
Not a problem, thank you @mostafaberg ! 👍

philips77, 2016-11-18:
Hi, the Version 3 is now published on our Version3 branch. You can edit your Profile and give the whole repo and branch path there. This bug has been fixed in this version, I hope :)
Version 3 is now in beta phase and we will publish it when we get some feedback.
Thanks!

jakerockland, 2016-12-08:
Version 3 seems to be working for us! Going to close this issue for now. 😄

closed 2016-12-08 by jakerockland