nordicsemi/IOS-DFU-Library #43: DFU state sticks on `.disconnecting`

jakerockland, 2016-10-25:
When the DFU process exits with error 302, it seems to stay perpetually on the `.disconnecting` state. I've tried calling the `abort()` controller method in my `didErrorOccur()` implementation, but am having a different issue there which I noted in #42

jakerockland, 2016-10-28:
@mostafaberg and this as well please? I think it may be connected to #42 ?

mostafaberg, 2016-10-31:
Looking into that, this is only happening on Legacy DFU i assume right ?

jakerockland, 2016-11-03:
I'm a bit confused with the distinction here when you say Legacy DFU vs Secure DFU, as I noted in #44 We're performing DFU with a secure firmware packet though.

jakerockland, 2016-11-03:
Does that clear things up at all @mostafaberg ?

mostafaberg, 2016-11-03:
Secure DFU is the latest vesion to our DFU library, it uses signed distribution files and is a completely different protocol, from our DFU Library side it doesn't matter at all, the library understands which DFU type it's dealing with and handles it with the proper classes, they are two completely different systems though, you can ask around in our devzone if you want to know the specifics of Secure DFU vs. Legacy DFU

jakerockland, 2016-11-03:
Ah I think I was just misunderstanding your question then. From the perspective of the DFU Library, it is all one thing right?

We're seeing this issue with the Secure DFU then though to answer your question, using the recent DFU library and secure packets on our firmware end.

mostafaberg, 2016-11-03:
Yeah I seem to have used the wrong words in my question, from the iOSDFULibrary's side, nothing changes at all, it all depends on your firmware's implementation, I was actually asking about your firmware and not the DFULibrary, you can't choose that in the DFU Library since it's automatically done :)

thanks for clearing it up for me 👍

philips77, 2016-11-18:
Hi, the Version 3 is now published on our Version3 branch. You can edit your Profile and give the whole repo and branch path there. Could you try with this version?
Version 3 is now in beta phase and we will publish it when we get some feedback.
Thanks!

jakerockland, 2016-12-08:
Version 3 seems to be working for us! Going to close this issue for now. 😄

closed 2016-12-08 by jakerockland