nordicsemi/IOS-DFU-Library #44: DFU state is `.completed` twice in a row

jakerockland, 2016-10-28:
@mostafaberg I'm not sure if this also might be connected to #43 and #42, but when completing the DFU, I'm seeing the state change to `.completed` twice in succession, is this expected behavior or a bug?

I've included a some print debug statements to give a better idea of what's going on:

```
DFU Progress: 99% completed.
DFU Progress: 99% completed.
DFU Progress: 99% completed.
DFU Progress: 99% completed.
DFU Progress: 100% completed.
DFU State: Completed
Cleanup performed successfully.
DFU State: Completed
Error: “firmware.zip” couldn’t be removed.
```

Where our state change handler looks something like:

```
    /**
     Handles changes in the DFU state, called whenever the state of the DFU Service has changed
     
     - Parameter state: The new current DFU state
     */
    func didStateChangedTo(_ state: DFUState) {
        #if DEBUG
            print("DFU State: \(state.description())")
        #endif
        switch state {
        case .completed:
            // Delete temporarily stored firmware ZIP
            FirmwareUpdate.cleanup(localFirmwarePath!)
        case .connecting, .starting, .enablingDfuMode, .uploading, .validating, .disconnecting: break
        case .aborted, .signatureMismatch, .operationNotPermitted, .failed: break
        }
    }
```

jakerockland, 2016-10-28:
@mostafaberg I'm not sure if this is just an issue of me misunderstanding how the SDK is supposed to behave.

jakerockland, 2016-10-28:
Would very much appreciate the input 😄

mostafaberg, 2016-10-31:
interesting find, this doesn't happen with the Legacy DFU right ?

mostafaberg, 2016-10-31:
I can't reproduce that on a Secure DFU process so I'm waiting for your confirmation

jakerockland, 2016-11-03:
@mostafaberg A bit confused actually by the Legacy DFU / Secure DFU discrepancy. For all of these we are using secure DFU packets on our firmware end. As far as I understood, with the new SDK there wasn't a change we had to make to use the Secure DFU over the Legacy DFU, but that may just be something that is unclear to me and I'm going about things wrong.

When I initiate the DFU I do so like this:

```
let selectedFirmware = DFUFirmware(urlToZipFile:firmwareURL)
let initiator = DFUServiceInitiator(centralManager: self.manager.central, target: self.manager.peripheral!).withFirmwareFile(selectedFirmware!)
            
initiator.logger = self // To get log info
initiator.delegate = self // To be informed about current state and errors
initiator.progressDelegate = self // To show progress bar

self.controller = initiator.start()
```

mostafaberg, 2016-11-03:
sorry for not updating you earlier, but my colleague reproduced this issue "intermittently depending on which app is implementing the library", i'll try to trace it down with him and have a fix out ASAP !, thanks for all your input!

jakerockland, 2016-11-03:
Not a problem, thanks so much for looking into this @mostafaberg !

mostafaberg, 2016-11-03:
No worries, will keep you updated here!, if you have any more findings in the mean time just update the thread

jakerockland, 2016-11-03:
Awesome, thank you @mostafaberg !

philips77, 2016-11-15:
Hi, there will be an update to the lib soon. The double .completed will be fixed. Also expect some small changes in the API (delegate method has been renamed, SecureDFUState and ..Errors removed, perhaps the peripheralSelector will be slightly modified). But all should be easily fixable.

philips77, 2016-11-18:
Hi, the Version 3 is now published on our Version3 branch. You can edit your Profile and give the whole repo and branch path there. This bug has been fixed in this version.
Version 3 is now in beta phase and we will publish it when we get some feedback.
Thanks!

jakerockland, 2016-12-08:
Version 3 seems to be working for us! 😄

jakerockland, 2016-12-08:
Going to close this issue for now. :smile:

closed 2016-12-08 by jakerockland