nordicsemi/IOS-DFU-Library #45: Connected peripheral doesn't start DFU

kklobe, 2016-11-03:
I've tested 2.1.6 after seeing issue #23 closed and still can't get a connected peripheral to work. The issue seems to be that I've already discovered the services in my app after connecting, so the discoverServices call in DFUExecutor doesn't discover anything. Maybe the assumption should that the user has already done service discovery on a connected device, and iterate through peripheral.services just like the code in the didDiscoverServices callback? (I've hacked the code locally to do just that).

Thanks!

mostafaberg, 2016-11-03:
Thanks for the detailed feedback, this will be very helpful squash that bug !

I'll keep you updated and i've reopened #23 !

philips77, 2016-11-15:
Hi, there will be an update to the lib soon. We'll make sure the connected devices will also work. The service will do a service discovery if no dfu service is found (for example your app discovered only those that requires itself).

philips77, 2016-11-18:
Hi, the Version 3 is now published on our Version3 branch. You can edit your Profile and give the whole repo and branch path there. I hope it should work better now.
Versino 3 is now in beta phase and we will publish it when we get some feedback.
Thanks!

closed 2016-12-15 by philips77