nordicsemi/IOS-DFU-Library #46: Cannot include project into workspace without using Cocoapods

guidomb, 2016-11-03:
I tried to follow the docs but I cannot find the `DFULibrary.xcodeproj` file. Am I missing something? I don't use Cocoapods nor Carthage.

mostafaberg, 2016-11-03:
We do not generate any files named `DFULibrary.xcodeproj`, where have you read that in the docs ?

If you do not want to use Cocoapods you can use Carthage which will give you the compiled framework files directly

guidomb, 2016-11-04:
Check item 3 of this section in the docs https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/documentation.md#integration

I don't use Cocoapods. I do use Carthage but I use it with submodules, that is why I need the `.xcodeproj` file.

mostafaberg, 2016-11-04:
I'm so sorry that would be a mistake on our side with the documentation, this is no longer valid.
So what does submodules have to do with building the library ? shouldn't it simply work out of the box ?

The steps would be building with Carthage (you can do that outside your project directory), and simply copy the generated `iOSDFULibrary.framework` and `Zip.framework` files into your project as precompiled libraries, then you can update your submodules in the project normally and it should work

mostafaberg, 2016-11-04:
if you have any specific errors or more details on what's going wrong please provide them and we'll help you further

guidomb, 2016-11-04:
There usually 4 ways of adding dependencies to an iOS project

1. Using Cocoapods.
2. Using Carthage to build a `.framework`.
3. Using Carthage to manage git submodules.
4. Adding the `.xcodeproj` file to your project's workspace.

There is also a way to use Carthage as a "version resolver" and then tell carthage to configure a git submodule for each dependency that appears in the `Cartfile.resolved` file. If you go with this option then you need to that the `.xcodeproj` file of each dependency inside the `Carthage/Checkout` folder to your project's workspace. For more information about Carthage and git submodules check their [docs](https://github.com/Carthage/Carthage#using-submodules-for-dependencies).

For option 3 and 4 to work a `.xcodeproj` needs to be provided. Being able to add dependencies using git submodules has the advantage of being able to debug the source code when you run your project.

Most iOS libraries provide this way of adding them to other projects. For example take a look at [MagicalRecord](https://github.com/magicalpanda/MagicalRecord/blob/master/Docs/Installing-MagicalRecord.md#using-an-xcode-subproject).

mostafaberg, 2016-11-04:
Unfortunately we do not support those methods right now and i'm not sure about future support due to time restrictions, I'll look into it and evaluate what needs to be done, if it's a simple modification to our project's scheme or something of that sort i'll get it done soon !, will keep you updated.

but for the time being if you want to be able to step into our code and debug, the only way that works now is using `Cocoapods`, hope that you can work it out until we see what can be done there!

guidomb, 2016-11-04:
For now I copy pasted the library code and everything worked well. If you want to add this feature you only need to create an xcode project with framework target and that is it. Let me know if you need any help with that.

mostafaberg, 2016-11-04:
Ok I'll be looking into that and will come back to you next week, glad to hear it worked out and that it's a simple fix to get it up and running !

mostafaberg, 2016-11-18:
Hey!, I'm looking into his today, can you please clarify a bit more on your recommendation ?

guidomb, 2016-11-18:
Sure. I would suggest creating a new Xcode project from scratch. Create a framework project for iOS. Make sure that your scheme is shared (this is necessary for carthage). You can then add all the source files to that new project and that is pretty much it. You can check that everything is working by running `carthage build --no-skip-current`. You may need to upate the `.podspec` file if you change the source code location.

philips77, 2016-11-18:
Hi, btw, I've just released version 3 of DFU on https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/tree/Version3. Could you please start using this one? I have no idea how to add it to carthage, etc... API slightly changed (methods were renamed, all described in the documentation.md)

guidomb, 2016-11-18:
OK if you want I can give it a try and try to fix this issue. @philips77 Should I base my fix on the `Version3` branch to avoid the least amount of conflicts possible?

philips77, 2016-11-19:
Yes, the Version3 branch will be merged into master when tested more. Regarding Carthage or Pods @mostafaberg knows much more so he will take from here.

mostafaberg, 2016-11-21:
@guidomb if you do have time to do a PR with a fix that suits your needs we will surely take that!
i'm going to be taking another look today after i'm done with some other tasks, the main issues i found yesterday was making it build without having the `Zip` dependency that's supplied by the bundled cocoapods installation, but I didn't continue yet

guidomb, 2016-11-21:
@mostafaberg I'll try to find sometime this week. In the mind time I came across this tool which can help generating projects with support for Carthage, CocoaPods, SwiftPackagaManager and it also generates an XCode project -> https://github.com/JohnSundell/SwiftPlate

guidomb, 2016-11-29:
Here is another tool to generate a project structure that has support for all mayor dependency manager and generates the appropriate Xcode project that can be included in any workspace. May you can give it a try, generate a new project, copy paste source files to `Source` folder and update `.podspec` file to be up-to-date.

https://github.com/cookiecutter-swift/Framework

mostafaberg, 2017-08-09:
I have a quick Feature branch that I'd like you to test, this adds the feature you requested, please let me know if it works or not :), you may find it on my comment [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/57#issuecomment-321272560)

closed 2019-01-29 by philips77