nordicsemi/IOS-DFU-Library #47: Re-trying DFU begins before `initiator.start()` is called

jakerockland, 2016-11-03:
When attempting to perform a DFU, if the DFU fails for the first time for whatever reason, when trying to re-perform the DFU, it immediately begins upon connection to our peripheral, before the `initiator.start()` method had even been called.

Our current setup is that we have a button that initiates the DFU process. Pressing this button should initiate a connection to our peripheral, once connected to our peripheral there is some setup code that is run which retrieves the appropriate .zip file from a server of ours (this is all in our `centralManager(_ central: CBCentralManager, didConnect peripheral: CBPeripheral)` method), and once that is retrieved we initiate the DFU as follows:

```
let selectedFirmware = DFUFirmware(urlToZipFile:firmwareURL)
let initiator = DFUServiceInitiator(centralManager: self.manager.central, target: self.manager.peripheral!).withFirmwareFile(selectedFirmware!)
                
initiator.logger = self // To get log info
initiator.delegate = self // To be informed about current state and errors
initiator.progressDelegate = self // To show progress bar

self.controller = initiator.start()
```

This all works as expected the first time we attempt the DFU; however if the DFU fails and we try to re-attempt performing the DFU (by pressing our button again), as soon as we connect to our peripheral with the CBCentralManager `.connect()` method, the DFU process begins again, bypassing our `centralManager(_ central: CBCentralManager, didConnect peripheral: CBPeripheral)` method and preventing our necessary setup code from being called. 

Is there something here that we're doing wrong @mostafaberg ? I'm not sure if I may just be misunderstanding the expected behaviour of the library or if this is a bug.
 
I'm not sure if this is connected at all to the other recent issues I had opened, or potentially just an issue with how I'm using the DFU library but would really appreciate any input that could help explain why we're seeing this behavior.

mostafaberg, 2016-11-03:
Are you using the library without any modifications ? or did you do any modifications to the library's code?

If not i would guess it depends on where the DFU process failed and why, if it fails after sending the first part of a multiple part flash file (SD/BL+APP) etc.. it should automatically reconnect and resume

So it mostly depends on your implementation and the reason things are failing.

hard to tell without seeing some code and what is exactly triggering.

The first thing i would try without seeing your code is taking back the delegate of the central manager upon failure or before even connecting to the peripheral.

jakerockland, 2016-11-03:
Thanks for the feedback @mostafaberg 👍 

We're using the library without modifications (currently using 2.1.4 but I will update to 2.1.6 soon). We're currently seeing this behaviour when DFU fails with 202 error. 

Your comments led me in the right direction for a solution though, ended up needing to add a line to our error handler that sets the delegate for our central manager back to its default behaviour, which I've included a snippet of below (self.manager is the singleton that we're using to manage our Bluetooth connections):

```
 func didErrorOccur(_ error:DFUError, withMessage message:String) {
    // Other error handling stuff is here

    // Reset delegate for central manager
    self.manager.central.delegate = self.manager
}
```

Is this something that should be done when the DFU process completes normally as well, or only in the case of an error? I'm a bit unclear as to what is happening on the DFU Library end here.

philips77, 2016-11-15:
Hi, yes. The library takes over the delegates of centralManager and peripheral and never sets then back.
There will be an update to the library soon but most likely still delegates won't be restored.

jakerockland, 2016-11-15:
@philips77 I think leaving it to have the user of the library re-set the delegate makes sense, but perhaps this is something that could be better documented in the new update to the library? This was a tricky thing to debug and figure out 👍 😛

jakerockland, 2016-11-15:
@mostafaberg @philips77 Going to mark this issue as closed for now then as the solution seems to be resolved now--returning the control of the delegate upon termination the DFU process (both because of regular completion or unexpected exiting/error handling).

closed 2016-11-15 by jakerockland

philips77, 2016-11-15:
We'll make sure it's highlighted somewhere in a footer with small letters ;)

jakerockland, 2016-11-15:
@philips77 😂 😂 😂 😭 😭 😭