nordicsemi/IOS-DFU-Library #48: Crash in LegacyDFUPeripheral's isInApplicationMode

casey-pax, 2016-11-15:
First of all, thank you to the team for responding so quickly to all the issues. I've been following the repo and it's incredibly helpful.

On unmodified v2.1.6 I received a crash in isInApplicationMode as the LegacyDFUPeripheral's dfuService is nil. I haven't seen this a lot. I might have had two peripherals in dfu mode at the same time judging by the logs.

LOG:
`DFU Log: Connected to DFU_ERA`
`DFU Log: Services discovered`
`DFU Log: DFU characteristics discovered`
`DFU Log: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):0800`
`DFU Log: Version number read: 0.`8
`DFU Log: Connected to DFU_ERA`
`DFU Log: DFU Control Point notifications enabled`
`fatal error: unexpectedly found nil while unwrapping an Optional value`
`fatal error: unexpectedly found nil while unwrapping an Optional value`

CALLSTACK:
`0x0000000100b891ec in specialized _fatalErrorMessage(StaticString, StaticString, StaticString, UInt, flags : UInt32) -> Never ()`
`0x0000000100888a0c in LegacyDFUPeripheral.isInApplicationMode(Bool) -> Bool at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Peripherals/LegacyDFUPeripheral.swift:150`
`0x00000001008847a8 in LegacyDFUExecutor.onControlPointEnabled() -> () at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/DFU/LegacyDFUExecutor.swift:96`
`0x00000001008868c8 in protocol witness for DFUPeripheralDelegate.onControlPointEnabled() -> () in conformance LegacyDFUExecutor ()`
`0x0000000100888608 in LegacyDFUPeripheral.(enableControlPoint() -> ()).(closure #1) at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Peripherals/LegacyDFUPeripheral.swift:136`
`0x00000001008635f4 in thunk ()`
`0x0000000100864fb0 in DFUControlPoint.peripheral(CBPeripheral, didUpdateNotificationStateFor : CBCharacteristic, error : Error?) -> () at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Characteristics/DFUControlPoint.swift:313`
`0x0000000100865098 in @objc DFUControlPoint.peripheral(CBPeripheral, didUpdateNotificationStateFor : CBCharacteristic, error : Error?) -> () ()`
`0x0000000191ee3364 in -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] ()`
`0x0000000191ee34c4 in -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] ()`
`0x0000000191edfef4 in -[CBPeripheral handleMsg:args:] ()`
`0x0000000191edb018 in -[CBCentralManager handleMsg:args:] ()`
`0x0000000191eeef58 in -[CBManager xpcConnectionDidReceiveMsg:args:] ()`
`0x0000000191ee770c in __30-[CBXpcConnection _handleMsg:]_block_invoke ()`
`0x00000001009d525c in _dispatch_call_block_and_release ()`
`0x00000001009d521c in _dispatch_client_callout ()`
`0x00000001009e2b54 in _dispatch_queue_serial_drain ()`

philips77, 2016-11-15:
Thanks for notice. I'll take a look tomorrow. 
There will be a bigger update in coming days and will ask you to test it again when ready.

philips77, 2016-11-15:
Are you trying to do dfu for 2 devices at the same time? We have never tried it... But as the peripheral object is different, and the centralManager's delegate is used only at the beginning, it may have some chances to work.
I'll check tomorrow why the dfuService may be nil.

casey-pax, 2016-11-15:
We only try to update one at a time. That being said, there are more than one advertising while in DFU mode. So the Nordic SDK might be picking up more than one?

philips77, 2016-11-15:
Only the peripheral given to the initiator will be chosen unless you update SD/SD+BL and App from one zip. Then it will scan for the new bootloader (use peripheralSelector to filter devices).

casey-pax, 2016-11-15:
Sure, I'm just shooting in the dark as to whether there may be a concurrency bug or something if two devices are broadcasting. If I find spare cycles, I'll do some testing and report back.

philips77, 2016-11-18:
Hi, I've just released version 3 on new branch. Could you try it? You need to specify the repo and branch name in you Podfile. I did not make any changes in isInApplicationMode, I think, but maybe it will work better.
The Version 3 is now in beta phase and will be pushed to Pods when we get some feedback).

philips77, 2016-12-13:
Hi @casey-pax, do you still have this issue with the new version of the library?

philips77, 2016-12-15:
I'm closing the issue as the new version of library is out now. If you still experience this problem, please open it again.

closed 2016-12-15 by philips77