nordicsemi/IOS-DFU-Library #49: Conflict with `Zip` dependency for Carthage users

jakerockland, 2016-11-15:
Though the Zip dependency has been somewhat handled by making it explicit in the Cartfile of the repo here, there is still a conflict between the Zip library that is checked out as a dependency and the one in the `Example/Pods` directory. Is it possible to remove the `Zip` directory there altogether? I don't use CocoaPods so I may not be fully understanding part of this, but isn't it something that should be checked out as a dependency and not committed directly to the library repo here?

mostafaberg, 2016-11-17:
Hi Jake!
Sorry for the inconvenience.
you are right about it being a dependency that needs to be installed later on, however, some projects, like ours for example, decide to track the pods installation so that anyone that checks it out is able to build out of the box without needing to install cocoapods.

I have managed to build it with carthage before without any issues.

can you give me the exact steps that you take so i can reproduce the issue ?

jakerockland, 2016-11-17:
Hey @mostafaberg this is how we have our Cartfile set up:
```
github "delba/Permission" ~> 2.0
github "NordicSemiconductor/IOS-Pods-DFU-Library" ~> 2.1
github "Alamofire/Alamofire" ~> 4.0
github "delba/Log" ~> 1.0
```

When I build our Carthage dependencies with `carthage update --platform iOS`, everything works well without any issue:
```*** Fetching Log
*** Fetching Alamofire
*** Fetching IOS-Pods-DFU-Library
*** Fetching Permission
*** Fetching Zip
*** Checking out Alamofire at "4.1.0"
*** Checking out Log at "v1.0"
*** Checking out Permission at "v2.0"
*** Checking out Zip at "0.6.0"
*** Checking out IOS-Pods-DFU-Library at "2.1.6"
*** xcodebuild output can be found in /var/folders/8s/04msnzh10gq4t4_7phkdhr100000gn/T/carthage-xcodebuild.GE8A4P.log
*** Building scheme "Alamofire iOS" in Alamofire.xcworkspace
*** Building scheme "Log iOS" in Log.xcodeproj
*** Building scheme "Permission" in Permission.xcodeproj
*** Building scheme "Zip" in Zip.xcodeproj
*** Building scheme "iOSDFULibrary" in iOSDFULibrary.xcworkspace
```

However, when I go to build our project, I get an issue because of a "Redefinition of module 'minizip'"

[image]

mostafaberg, 2016-11-17:
Is the Carthage directory inside the project's root?

jakerockland, 2016-11-17:
Yup, our current directory structure looks like this:

[image]

mostafaberg, 2016-11-17:
Yeah I guessed that was the issue, problem is you're now importing files that are not intended to be in there, you need to just use the contents of carthage's build directory and not the checkouts directory , you only need the framework files in the build dir in your project

Do you have a reason that requires having a carthage inside the project ?

jakerockland, 2016-11-17:
@mostafaberg It's definitely _possible_ to delete the `Checkouts` directory, as the built frameworks are all we need to build the project, which is why this isn't a fatal issue in any respect (this is what we've been doing). But since the `Checkouts` directory will be generated every time the Carthage dependencies are rebuilt with `carthage update --platform iOS`, it seems like unideal behavior to have to manually delete the `Checkouts` directory in order for this Framework to work. 😅

jakerockland, 2016-11-17:
I wanted to bring the issue up though as it seems like unideal behavior on the side of this Framework, when other Carthage Frameworks usually don't require you to delete the Checkouts folder, though perhaps we just want to mark this as a `wontfix` issue depending on if it is necessary to have the `Zip` files committed to the repo in the `Pods` directory or if there is a better work-around available there (I don't know enough about CocoaPods to offer any suggestions there 😓).

mostafaberg, 2016-11-17:
Removing the zip dependency in cocoa pods is not an option for us as then the user will have to install cocoapods to test the project.

Is the Carthage checkouts directory added to the xcode project ? Or are they only in the file system?

jakerockland, 2016-11-17:
@mostafaberg That makes sense as to why you need it included for that purpose then, hmm... The `Checkouts` directory is not added to the XCode project as far as I understand and is only in the file system, which is why its perplexing me a bit that the conflict arrises in the first place, perhaps I'm missing something here?

[image]

mostafaberg, 2016-11-18:
I agree that's weird indeed, maybe your search paths have some configuration that causes this ?

I'll actually try to reproduce this with Carthage within the project and see what happens and will update you

mostafaberg, 2016-11-18:
Ok issue reproduces here !, looking into a fix right now

mostafaberg, 2016-11-18:
Unfortunately I can't see a quick fix for this right now, the easiest thing i can suggest until we do a proper fix is to add a build step in your project that simply deletes the file `Pods/Zip/Zip/minizip/module.modulemap` before building or having the carthage build outside the project, in the same project root should be fine i think ?

AlexDM0, 2017-01-17:
Is this still being considered?

mostafaberg, 2017-01-17:
Hi @AlexDM0, Not really as this is a very low priority issue, we have concluded for now that it's best to build using carthage in the project's root or outside the project rot and manually importing the required files. will close this issue to avoid confusion, if we have a better solution in the future we will surely mention it in the change logs.

if you have a related issue or other questions just open a new issue and we can handle it from there

closed 2017-01-17 by mostafaberg

AlexDM0, 2017-01-17:
While my lib compiles using the run script 

```
# required to build framework in xcode
rm -f ./Carthage/Checkouts/Zip/Zip/minizip/module.modulemap
```

in the build phase, I also package my lib for use in carthage with other apps.

When I include this in another app, I need to include the minizip dependency manually and I need to add this line 

```
# required to build framework when used with carthage externally
rm -f ./Zip/Zip/minizip/module.modulemap
```

This removes the redifinition notice but I get another issue that the Minizip is not available. I added minizip to my carthage list but it again fails saying could not build objective C module minizip.

To be clear, this only happens when ios pods DFU lib is part of a lib that is included by carthage in another app.

Any ideas?

jakerockland, 2017-01-17:
@AlexDM0 I don't know if this is helpful to you, but this is what we do.

Add the `IOS-Pods-DFU-Library` to your Cartfile regularly:

`github "NordicSemiconductor/IOS-Pods-DFU-Library" ~> 3.0`

To build dependencies:

`carthage update --platform iOS`

To cleanup the `Zip` conflict:

`rm -rf Carthage/Checkouts/IOS-Pods-DFU-Library/Example/Pods/`

We use Fastlane, so we just have a lane set up for building dependencies, which looks like this:

```
  desc "Update dependencies"
  lane :dependencies do
      carthage(command: "update", use_binaries: false, platform: "iOS") # Builds all dependencies
      sh "bash ./dfulibcleanup" # Delete IOS-Pods-DFU-Library/Example/Pods directory
  end
```

mostafaberg, 2017-01-18:
@AlexDM0 please open a new issue, this is not the same bug

AlexDM0, 2017-01-18:
created #76

mostafaberg, 2017-01-18:
@jakerockland , @AlexDM0  Issue is now resolved in version `3.0.5` !
Remember to do the following:
1-Remove your build step / fastlane comand that did the cleanup in your project
2-Remove the `Zip` directory in your `Checkouts` directory to avoid future conflicts !

jakerockland, 2017-01-18:
@mostafaberg This worked for me, thank you!

mostafaberg, 2017-01-20:
Issue resolved in `3.0.6` by making the Zip framework scheme shared. now Carthage picks it up and builds both iOSDFULibrary and Zip frameworks!

give it a shot and let me know if you experience any more issues!

hiniesta, 2017-03-03:
Hi! I can run the App, but can't get the Archive work. I always get:

> strip -S Carthage/Build/iOS/Zip.framework.dSYM/Contents/Resources/DWARF/Zip -o Zip.framework.dSYM/Contents/Resources/DWARF/Zip
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/strip: string table not at the end of the file (can't be processed) in file: Carthage/Build/iOS/Zip.framework.dSYM/Contents/Resources/DWARF/Zip (for architecture i386)
error: strip Carthage/Build/iOS/Zip.framework.dSYM/Contents/Resources/DWARF/Zip: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/strip exited with 1