nordicsemi/IOS-DFU-Library #52: DFU  can not init a DFUFirmware

JJCoderiOS, 2016-11-28:
Hello, my introduction to upgrade your framework to my hardware in my own framework, but according to a zip file path to initialize the DFUFirmware has failed, and the use of non framework would not have this problem. The picture I have given you a screenshot, hope to reply, thank you!

JJCoderiOS, 2016-11-28:
i want to import your project into my sdk ,but it's always  initialize DFUFirmware faild

philips77, 2016-11-28:
Hi, the Zip must be generated using nrfutil tool (must have the correct manifest.json file, etc).
https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/README.md

philips77, 2016-11-28:
Also, could you please use the Version3 branch? It will replace the master soon but we would like to get a feedback before that. Thanks!

You can configure the Podfile like in here https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/master/Podfile

JJCoderiOS, 2016-11-28:
OK,thanks your reply,i will try again,later give you feedback

JJCoderiOS, 2016-11-28:
sorry,I did what you said, but  I still didn't succeed.
i think  the zip file  is No problem ,i use the zip file can success in other project(not sdk)

[image]

philips77, 2016-11-28:
Are you sure the url variable is not nil? Is your bin file word-aligned? (Bin_Size % 4 = 0)
You should be able to debug into this constructor if DFUFirmware, don't you?

PS. I hope you did "pods update" command and resolved method renaming issues?

mostafaberg, 2016-11-28:
What kind od url is it? Can you print the url path?

JJCoderiOS, 2016-11-28:
url path :  /var/mobile/Containers/Data/Application/A15AB70E-B110-4CD3-983A-CA4E09C722AB/Library/Caches/ble_app_wechat_debug_v1.1.zip

mostafaberg, 2016-11-28:
I think this is your problem, it's a file path not a file URL, try instantiating the fliepath variable this way instead:
```
NSURL *fileURL = [[NSBundle mainBundle] URLForResource:@"ble_app_wechat_debug_v1.1" withExtension:@"zip"];
DFUFirmware *firmware = [[DFUFirmware alloc] initWithUrlToZipFile:fileURL];
```

JJCoderiOS, 2016-11-28:
thanks your patience,  
now my fileUrl is:

file:///var/mobile/Containers/Data/Application/69D7C61F-69DE-4D75-8FFC-169F6CF9634C/Library/Caches/ble_app_wechat_debug_v1.1.zip  

my file is download from server ,then i save it ,  

like:

NSString *savePath = NSSearchPathForDirectoriesInDomains(NSCachesDirectory, NSUserDomainMask, YES).firstObject;
    NSString *filePath= [savePath  stringByAppendingPathComponent:@"ble_app_wechat_debug_v1.1.zip"];

then i get it from the filePath and instantiating it to url  use :    NSURL *fileURL = [NSURL fileURLWithPath:self.filePath];
then i use the  fileURL init the DFUFirmware 

you can get Correct results? don't have any problem?

mostafaberg, 2016-11-28:
No problems, I can't run it since i don't have your full project.

Where are you downloading your file ? you are probably referencing a file that doesn't exist, i think it's downloaded somewhere else, can you print it's path where it's being downloaded and verify it's the same place ?

JJCoderiOS, 2016-11-28:
the follwing is my zip  file(contail 5 files)  and the  manifest.json

[image]

[image]

mostafaberg, 2016-11-28:
Are you downloading the file and storing it into XCode ? or download it inside the app ?

JJCoderiOS, 2016-11-28:
inside tha app,  storing in  NSCachesDirectory

mostafaberg, 2016-11-28:
Can you post the code where you download the file ?

JJCoderiOS, 2016-11-28:
NSString *savePath = NSSearchPathForDirectoriesInDomains(NSCachesDirectory, NSUserDomainMask, YES).firstObject;
    
    NSOperationQueue *queue = [[NSOperationQueue alloc ]init];
    
    NSURL *url = [NSURL URLWithString:@"https://public.dingdingtingche.com/ddServerOffline/ble_app_wechat_11151139.zip"];

    NSString *filePath= [savePath  stringByAppendingPathComponent:@"ble_app_wechat_debug_v1.1.zip"];
    AFHTTPRequestOperation *op = [[AFHTTPRequestOperation alloc]initWithRequest:[NSURLRequest requestWithURL:url]];
    op.outputStream = [NSOutputStream outputStreamToFileAtPath:filePath append:NO];
    
    [op setDownloadProgressBlock:^(NSUInteger bytesRead, long long totalBytesRead, long long totalBytesExpectedToRead) {
        
        
        
    }];
    
    [op setCompletionBlockWithSuccess:^(AFHTTPRequestOperation *operation, id responseObject) {
        
        NSLog(@"%@",filePath);
                
    } failure:^(AFHTTPRequestOperation *operation, NSError *error) {
        
        NSLog(@" down faild  reasone is : %@",error);
        
    }];
    //start
    [queue addOperation:op];

mostafaberg, 2016-11-28:
The firmware file seems to be parsed correctly on my side, i'm pretty sure it's your download code that's causing an issue here.

can you check if the file exists in the success completion block ?

instead of `NSLog(@"%@",filePath);` you can have something like:

```
NSFileManager *fileManager = [NSFileManager defaultManager];
if (![fileManager fileExistsAtPath: filePath]) {
    NSLog("ERROR: File not found at %@", filePath);
} else {
   NSLog("File found at %@", filePath);
}
```

let me know what that shows

JJCoderiOS, 2016-11-28:
[image]

wo want to know what's the different in sdk and !sdk   i use the same code don't have problem in !sdk
but ,when i create a framework   then use it in framework  it will have the problem

mostafaberg, 2016-11-28:
What do yo mean by `sdk` and `!sdk` ? also, can you please show the actual output of the code in the console and not the screenshot ?

JJCoderiOS, 2016-11-28:
File found at /var/mobile/Containers/Data/Application/D6B3B215-3325-4068-8975-CB6880234BFF/Library/Caches/ble_app_wechat_debug_v1.1.zip

JJCoderiOS, 2016-11-28:
sdk means framework     !sdk means an empty project

mostafaberg, 2016-11-28:
ok so now we know at least the file is there in place, what are you doing in the `startWrite` method ?
I'm not sure what would be different between a framework and an empty project, it all depends on how you implement things, but it should be the same thing

icerainmm, 2016-11-28:
our app is developed based on our bleSDK(our base ble framework). at the first, we import DFUFramework directly in app, it works well. but when we import DFUFramework into our ble framework, and download the zip in app,  then pass zip filepath as argu into ble framework(which is imported with DFUFramework). here we met the above issue.

mostafaberg, 2016-11-28:
Aha i see, so is the code above from your own framework ? or is it form within the app ?

icerainmm, 2016-11-28:
download implement is in app, and filepath parse to URL & init DFUFirmware is in own framework. we pass download path when download is completed to own framework as argu.

mostafaberg, 2016-11-28:
try to pass a file URL and not a file path

mostafaberg, 2016-11-28:
as long as our `iOSDFULibrary` receives a file URL it should work, i think in your case your library is actually forwarding a file path as an URL not an actual file URL to `iOSDFULibrary`, that's why it works when used directly in the app

icerainmm, 2016-11-28:
thank you for your reply, we still stuck here after trying pass fileURL or put download implement in our own framework. 
we'll continue investigate this issue tomorrow. look forward your reply if you have any other idea.

icerainmm, 2016-11-29:
@mostafaberg thank you for your reply for the issue we met.

we finally solve above issue by cleaning whole project and re-import DFULibrary with the third way "Include the library as source code". even though we're not sure what is the root cause, anyway it is solved.

thanks a lot again.

JJCoderiOS, 2016-11-29:
thanks a lot 👍

closed 2016-11-29 by JJCoderiOS

mostafaberg, 2016-11-29:
Glad to hear you got things working, the problem was a bit strange and will try to look into it and see if there's something that's wrong on our side