nordicsemi/IOS-DFU-Library #53: Validation of archive fails at codesigning Zip.framework

jkestner, 2016-11-28:
Built 2.1.6 in Xcode 8.1 with "carthage build --platform iOS". Everything seems to work fine when running on my development devices. But when I try to put it up on iTunes Connect for my beta testers, it fails during validation, saying "Codesign failed" and "Unable to validate your application.
The archive is invalid. /var/folders/xx/blahblah/T/XcodeDistPipeline.5Lf/Packages/MyApp.ipa does not exist." It does this when the status is "Codesigning Zip.framework..."

New to Carthage so wondering if I missed something, but the copy-frameworks script does appear to be doing its job of stripping out unneeded architectures, and digging into the .ipa, the Zip binary does appear to be signed. Any ideas?

mostafaberg, 2016-11-29:
Could it be an issue related to that ? let us know if this helps!
https://github.com/Carthage/Carthage/issues/799

jkestner, 2016-11-29:
It doesn't appear to be a bitcode problem. No errors for that.
I removed the frameworks and still encountered the same problem with my bare app where it was previously fine. Found an "no identity found" error that I hadn't noticed before, and eventually had to turn off "Automatically manage signing" and get new certs. Ugh, but it'll work for now. This is the first time it's happened so I still want to blame Carthage or the addition of a Swift-based library for messing up the codesigning somehow, but no proof.

closed 2016-11-29 by jkestner