nordicsemi/IOS-DFU-Library #54: Error when Dfu_v1 firmware is greater than 0xFFFF bytes

sschiffli, 2016-11-30:
Copy from https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/25

The exact error we are seeing is "Error: 65760 bytes were sent while 224 bytes were reported as received"

The issues stems from the fact that older bootloader versions represented the received firmware image size as a uint16_t (thus we saturate) and sometime later it changed to uint32_t. This library always interprets the field as a uint32_t. https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/master/DFULibrary/DFUControlPoint.swift#L196

Since we still have old devices in the field to support I've had to modify [LegacyDFUService.swift](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift#L333) from
`if bytesSent == bytesReceived {`
to
`if bytesSent == bytesReceived || bytesSent == (bytesReceived + 0x10000) {`

philips77, 2016-11-30:
Hi, could you try Version3 branch? 
You can change the Podfile as in nRF Toolbox app.

philips77, 2016-11-30:
Ok, I think I have seen this issue on Android Dfu lib. I'll take a look tomorrow. The fix will go to the version3 branch.

philips77, 2016-11-30:
We also accept pull requests :)

sschiffli, 2016-12-01:
Still present on Version3, I'll submit my workaround in a pull request

mostafaberg, 2016-12-01:
Thanks a lot for the PR ! we will look into it and Merge as soon as possible :)

philips77, 2016-12-01:
Hi @sschiffli,
Could you try now with the Version3 branch? If you are using Pods, and configured it to use the Version3 branch, just type ```pods update``` in console and the new commits should be obtained. I did some tests but could you verify it works in your project?
Thanks again for reporting the issue!

sschiffli, 2016-12-01:
Looks great, thanks!

sschiffli, 2016-12-01:
eb83378 broke the case where the legacy DFU _did_ report bytes received as 32bits (sometime after the SDK 5.2.0.39364 release).
`if (bytesSent & 0xFFFF) == bytesReceived {`
Once bytes received goes over 0xFFFF this comparison will fail.  I think we should mask both bytesSent and bytesReceived.
`if (bytesSent & 0xFFFF) == (bytesReceived & 0xFFFF) {`

philips77, 2016-12-01:
True. Good catch!

philips77, 2016-12-01:
Done.

closed 2016-12-01 by philips77

sschiffli, 2016-12-01:
I also noticed some changes in DFUControlPoint.swift which aren't really necessary.  Even in the  SDK 5.2.0.39364 DFU code, the response over the wire was in fact 5 bytes long.  This means all the code in the [else block](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/eb83378fba47d1400d5912df01d3f18b9a887f55/iOSDFULibrary/Classes/Implementation/LegacyDFU/Characteristics/DFUControlPoint.swift#L196) in unreachable.

philips77, 2016-12-01:
I was wondering why the app didn't crash before the bytesSent and bytesReceived comparison in your case... But I think we can leave it like that, hym?

sschiffli, 2016-12-01:
Whatever you think will be less confusing down the line from a maintenance perspective.  I just doubled checked to confirm that it is in fact dead code.

philips77, 2016-12-01:
Thanks again for testing it.

philips77, 2016-12-02:
Hi, after all I decided to remove the dead code and just add some information about the issue.
I pushed the https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/commit/0e2983c6252d9047cb4c4a0d29bcd3002f1031f0 to Version3 branch.

StrawHatAaron, 2018-09-02:
I am facing this issue with the cocoa pods version 'iOSDFULibrary', '~> 4.0'

except I get Error: 1080 bytes were sent while 40 bytes were reported as received"