nordicsemi/IOS-DFU-Library #55: forceDfu parameter not copied in DFUExecutor

sschiffli, 2016-11-30:
The generic executor doesn't copy the the forceDfu parameter from the original initiator.

I think that in both [startSecureDFU](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUExecutor.swift#L67) and [startLegacyDFU](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUExecutor.swift#L81) then should also have:
`dfuInitiator.forceDfu = initiator.forceDfu`

philips77, 2016-11-30:
Could you try the Version3 branch? This version is currently almost ready to replace the current master one.

sschiffli, 2016-12-01:
Issue solved on Version3, Thanks!

closed 2016-12-01 by sschiffli