nordicsemi/IOS-DFU-Library #58: DFU 'Reading DFU Version characteristic failed' with 'The handle is invalid' Error

sss989870, 2016-12-05:
(i'm korean, im can't speak english. please consider that...)
im using iOSDFULibrary with my company BLE and Firmware.
it run on Android DFULibrary, nrf Toolbox very well.
but on ios, not working DFULibrary and nrf Toolbox.
i don't know why

that's my Log. Using iOSDFULibrary

connecting
V : Discovering all services for peripheral <CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
V : Discovered Service 9FD41000-E46F-7C9A-57B1-2DA365E18FA1 on peripheral <CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
V : Discovered Service 9FD42000-E46F-7C9A-57B1-2DA365E18FA1 on peripheral <CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
V : Discovered Service 9FD43000-E46F-7C9A-57B1-2DA365E18FA1 on peripheral <CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
V : Discovered Service 00001530-1212-EFDE-1523-785FEABCD123 on peripheral <CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
V : Did discover legacy DFU service
V : Starting legacy DFU Service initiator
V : Instantiated Legacy DFU peripheral selector
V : Connecting to DfuTarg...
D : centralManager.connectPeripheral(peripheral, options:nil)
V : Started Legacy DFU service controller
connecting
D : [Callback] Central Manager did connect peripheral
I : Connected to DfuTarg
V : Discovering services...
D : peripheral.discoverServices(nil)
I : Services discovered
V : DFU Service found
V : Discovering characteristics in DFU Service...
D : peripheral.discoverCharacteristics(nil, forService:DFUService)
I : DFU characteristics discovered
D : <CBCharacteristic: 0x1740b2e40, UUID = 00001532-1212-EFDE-1523-785FEABCD123, properties = 0x4, value = (null), notifying = NO>
D : DFUPacket
D : <CBCharacteristic: 0x1700b9fe0, UUID = 00001531-1212-EFDE-1523-785FEABCD123, properties = 0x18, value = (null), notifying = NO>
D : DFUControlPoint
D : <CBCharacteristic: 0x1700b9f80, UUID = 00001534-1212-EFDE-1523-785FEABCD123, properties = 0x2, value = <0100>, notifying = NO>
D : DFUVersion
V : Reading DFU Version number...
D : peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)
D : <CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
D : <CBCharacteristic: 0x1700b9f80, UUID = 00001534-1212-EFDE-1523-785FEABCD123, properties = 0x2, value = <0100>, notifying = NO>
<CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
<CBCharacteristic: 0x1700b9f80, UUID = 00001534-1212-EFDE-1523-785FEABCD123, properties = 0x2, value = <0100>, notifying = NO>
nil
I : Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):0100
A : Version number read: 0.1
V : Enabling notifiactions for 00001531-1212-EFDE-1523-785FEABCD123...
D : peripheral.setNotifyValue(true, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123)
starting
V : Notifications enabled for 00001534-1212-EFDE-1523-785FEABCD123
A : DFU Control Point notifications enabled
W : Application with buttonless update found
V : Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D : peripheral.writeValue(0x0104, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
enablingDfuMode
I : Data written to 00001531-1212-EFDE-1523-785FEABCD123
A : Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
Optional(Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.})
D : The specified device has disconnected from us.
D : [Callback] Central Manager did disconnect peripheral
I : Disconnected by the remote device
jumpingToBootloader
V : Connecting to DfuTarg...
D : centralManager.connectPeripheral(peripheral, options:nil)
D : [Callback] Central Manager did connect peripheral
I : Connected to DfuTarg
V : Discovering services...
D : peripheral.discoverServices(nil)
I : Services discovered
V : DFU Service found
V : Discovering characteristics in DFU Service...
D : peripheral.discoverCharacteristics(nil, forService:DFUService)
I : DFU characteristics discovered
D : <CBCharacteristic: 0x1700b9860, UUID = 00001532-1212-EFDE-1523-785FEABCD123, properties = 0x4, value = (null), notifying = NO>
D : DFUPacket
D : <CBCharacteristic: 0x1700b4040, UUID = 00001531-1212-EFDE-1523-785FEABCD123, properties = 0x18, value = (null), notifying = NO>
D : DFUControlPoint
D : <CBCharacteristic: 0x1700b9f20, UUID = 00001534-1212-EFDE-1523-785FEABCD123, properties = 0x2, value = <0100>, notifying = NO>
D : DFUVersion
V : Reading DFU Version number...
D : peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)
D : <CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
D : <CBCharacteristic: 0x1700b9f20, UUID = 00001534-1212-EFDE-1523-785FEABCD123, properties = 0x2, value = <0100>, notifying = NO>
<CBPeripheral: 0x1740ff600, identifier = 864BAD24-5F97-4E02-96E1-AF130E2E0325, name = DfuTarg, state = connected>
<CBCharacteristic: 0x1700b9f20, UUID = 00001534-1212-EFDE-1523-785FEABCD123, properties = 0x2, value = <0100>, notifying = NO>
Optional(Error Domain=CBATTErrorDomain Code=1 "The handle is invalid." UserInfo={NSLocalizedDescription=The handle is invalid.})
E : Reading DFU Version characteristic failed
Optional(2 bytes)
E : Error 1: The handle is invalid.
E : The handle is invalid.
V : Disconnecting...
D : centralManager.cancelPeripheralConnection(peripheral)
D : [Callback] Central Manager did disconnect peripheral without error
I : Disconnected
didErrorOccur: Reading DFU Version characteristic failed
E : Reading DFU Version characteristic failed

philips77, 2016-12-05:
Hi, it looks like the services has been cached on ios.
If your device ID not bondable, make sure the Service Changed characteristic is present in the Generic Attribute service. Then iOS will not cache services at all.
If it's bondable, make sure you send Service Changed indication after switching to bootloader mode (sdk 8-11).
Also, some minor advices:
- use Version3 branch of this library. It will soon replace the master branch. See nRF Toolbox's Podfile how to configure it. Some small changes in api were made.
- use DFUServiceInitiator it start dfu. I don't know if you are starting it with the legacy version, or is it just something that is on the log.

sss989870, 2016-12-05:
thanks for considering it. :)
my device run on legacy version.. and sdk 9.0.0.
i dont know 'bondable' mean

'services has been cached' mean problem in project? or device?

philips77, 2016-12-05:
A bondable device is one that you have to bond (pair) to use it. After pairing the communication is encrypted. When bonded, you will see such device listed like here:
http://dl.waspbarcode.com/wasp/supportfiles/kbimages/scanners/wws550i/ios-pairing-2-before-pairing.gif

A non-bondable is one that can be used by anyone, without pairing. The communication is not encrypted.

'services has been cached' is not a problem, it's just how it is.
When a phone connects to a device for the first time it does service discovery, that means it asks it for services and characteristics. In order not to send the same data on the second+ connections the services are cached on the phone. This saves time and battery, but prevents from changing the attribute table (list of services) on the device.
As I wrote above there are 3 ways of dealing with it. In 2 forst cases, if you plan to change services on your device (using a DFU requires it), you have to have Service Changed characteristic. On non-bonded device this is enough. iOS will never cache services if this characteristic is found in the Generic Attribute service. On bonded devices after changing services your device needs to send the SC indication. Then the central (iOS) will invalidate cache and do the service discovery again.
The third method requires you to change device address. For example, after switching to DFU bootloader mode your device needs to change its MAC address (e.g. by adding 1). Then, it is visible as a new device for the phone and a new service discovery is done as it is a new device.

To check if your device has Service Changed characteristic connect to it using nRF Connect on Android or PC and you look for Generic Attribute service.

sss989870, 2016-12-05:
I will follow your comments very slowly and tell you the results.

my phone not bonded device, Different this picture 'http://dl.waspbarcode.com/wasp/supportfiles/kbimages/scanners/wws550i/ios-pairing-2-before-pairing.gif'

and i understanded your 'services has been cached' meaning explanation.

I check i have 'Service Changed characteristic' using nRF Connect for Android
In Generic Attrubute not has Service Changed Characteristic.
But Server has Service Changed Characteristic.

nRF Connect Service Image
Client : 'https://drive.google.com/open?id=0B3Lv_uliEGSjaDVSRTBJM3h5Zkk'
Server : 'https://drive.google.com/open?id=0B3Lv_uliEGSjV0FPbWprYVhQREk'

you say ' iOS will never cache services if this characteristic is found in the Generic Attribute service.'
then, iOS will cached services.

Is that right?

Thanks for the detailed explanation.

philips77, 2016-12-05:
Yes, exactly. In our SDK you just have to set the
```#define IS_SRVC_CHANGED_CHARACT_PRESENT 1``` 
flag to 1 in main.c file of your project and compile again.

sss989870, 2016-12-05:
Thanks you!

tomorrow I will try all and close issue!

closed 2016-12-13 by sss989870

NikhilDevgire, 2017-04-12:
Hi @DHyuk,

Are you able to solve your issue as I am still struggling with the same issue? If you found out any solution could you please provide any help.