nordicsemi/IOS-DFU-Library #61: Descriptor write in Ble OR How do you create a descriptor for a mutable characteristic?

ravindrarathore, 2016-12-07:
```
var parameter = NSInteger(01)
                let data1 = NSData(bytes: &parameter, length: 2)
              
                 let yourCharUUID:CBUUID = CBUUID(string:"8E400001-F315-4F60-9FB8-838830DAEA50")
                let yourCharacterstics : CBMutableCharacteristic = CBMutableCharacteristic(type: yourCharUUID, properties: CBCharacteristicProperties.writeWithoutResponse, value: nil, permissions: CBAttributePermissions.writeable)
                //let UserDescriptionUUid : CBUUID = CBUUID(string:"00002902-0000-1000-8000-00805f9b34fb")
                let USerDescriptionUUid : CBUUID =  CBUUID(string:CBUUIDClientCharacteristicConfigurationString)
                
                let YourDescriptor : CBMutableDescriptor = CBMutableDescriptor(type: USerDescriptionUUid, value:"myDescriptorValue" )
                yourCharacterstics.descriptors = [YourDescriptor]
                
                
                 peripheral.writeValue(data1 as Data, for: characteristic, type: CBCharacteristicWriteType.withResponse)

```
Getting Issue  - 1. Assertion failure in -[CBMutableDescriptor initWithType:value:], /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreBluetooth/CoreBluetooth-351/CoreBluetooth/CBDescriptor.m:80

2. Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'UUID "Client Characteristic Configuration" is not allowed'
*** First throw call stack:

mostafaberg, 2016-12-07:
You can't use mutable types for a given peripheral, this is only intended for server use, you need to first discover the characteristics and descriptors for the given peripheral and then find which descriptor you want to write to, then hold a reference to that descriptor, descriptors, characteristics and services for a CBPeripheral are not mutable and are readonly so you cannot instantiate your own and write to it.

1-Get the peripheral
2-discover all services and store them
3-discover characteristics in for each service and store them
4-discover descriptors for each characteristic and store them
5-Find the required descriptor from the discovered ones, then write to it

ravindrarathore, 2016-12-07:
Thanx
mostafaberg
 Actually i am doing
1-Get the peripheral
2-discover all services
3-discover characteristics in for each service 
4-discover descriptors for each characteristic 
in step 4  i get discover descriptors are nil for each charcterstics.

Bascially I want to send in bootloaderMode (Notify Descriptor and write characterstics 0x01 so device go in bootloader mode)
so please give any idea or example in which you send device in bootloader Mode .

philips77, 2016-12-07:
Hi, since version 3 jumping to bootloader mode is handled automation by the library in both legacy and secure dfu. For secure DFU the experimental buttonless service is required and you have to set a special flag in the DFUServiceInitiator. Do you need something else?

ravindrarathore, 2016-12-07:
Hi philips77 
yes i want  experimental buttonless service.
Whenever  i use  Cocoapods podfile install the iosDfu Library I getting error  (Stuck in  conversion swift 3 and conversion fail ). so  i use Include the library as a precompiled framework on 5 dec 2016.
so i need to Update library or Install Podfile ?

ravindrarathore, 2016-12-08:
Thanx man  For your Suggestion and Contribution Its works for me after updating podfile .

closed 2016-12-08 by ravindrarathore

ravindrarathore, 2016-12-08:
Thanx man my issue is solve after updating podfile.
Thanks man