nordicsemi/IOS-DFU-Library #62: Version3 doesn't retry

casey-pax, 2016-12-12:
If the DFU process was interrupted due to a disconnect, Version2 would retry the process.

Version3 doesn't seem to do so. Is it now up to us to re-scan and re-initiate?

philips77, 2016-12-13:
Hi,
Let us take a look into the code and discuss your issue. My or my college will come back will come back to you soon.

casey-pax, 2016-12-13:
Thanks. I will say that it seems to be much more reliable in giving a disconnect error than Version2.

If you decide to implement a retry, it would be useful to add that as a DFUState or an additional delegate method so we can display the appropriate UI. In Version2, I had to parse one of the error messages which was less than ideal :)

larsblumberg, 2017-01-02:
@casey-pax I implemented a workaround for this issue, I discussed it here: #71 

```
let dfuInitiator = DFUServiceInitiator(...)
...
var dfuController = dfuInitiator.start()
...
public func dfuError(_ error: DFUError, didOccurWithMessage message: String) {
    if error == .deviceDisconnected {
        dfuController.abort()
        dfuController = dfuInitiator.start()
    }
}
```

mostafaberg, 2017-02-07:
This issue has seemed to be resolved in 3.0.4, will close and please reopen a new issue if problem persists

closed 2017-02-07 by mostafaberg