nordicsemi/IOS-DFU-Library #63: Disconnect after writing image sizes, with iOS 10.1.1

starblanket, 2016-12-13:
I'm using iOS 10.1.1 on an iPhone 7 and the legacy DFU on my device. I have an Objective C project that's using version 3.0.0 of this library. 

The device disconnects unexpectedly after writing the images sizes:

```
OTAViewController.m -[OTAViewController installFirmware]:166 parts=1 softdevice=0 bootloader=0 application=63680
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Connecting to GaitReminder...
OTAViewController.m -[OTAViewController logWith:message:]:394 00 centralManager.connect(peripheral, options:nil)
OTAViewController.m -[OTAViewController dfuStateDidChangeTo:]:271 DFU state change to 0
OTAViewController.m -[OTAViewController logWith:message:]:394 00 [Callback] Central Manager did connect peripheral
OTAViewController.m -[OTAViewController logWith:message:]:394 05 Connected to GaitReminder
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Discovering services...
OTAViewController.m -[OTAViewController logWith:message:]:394 00 peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123, FE59])
OTAViewController.m -[OTAViewController logWith:message:]:394 05 Services discovered
OTAViewController.m -[OTAViewController logWith:message:]:394 05 Connected to GaitReminder
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Legacy DFU Service found
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Discovering characteristics in DFU Service...
OTAViewController.m -[OTAViewController logWith:message:]:394 00 peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
OTAViewController.m -[OTAViewController logWith:message:]:394 05 DFU characteristics discovered
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
OTAViewController.m -[OTAViewController logWith:message:]:394 00 peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
OTAViewController.m -[OTAViewController dfuStateDidChangeTo:]:271 DFU state change to 1
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
OTAViewController.m -[OTAViewController logWith:message:]:394 10 DFU Control Point notifications enabled
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
OTAViewController.m -[OTAViewController logWith:message:]:394 00 peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
OTAViewController.m -[OTAViewController logWith:message:]:394 01 Writing image sizes (0b, 0b, 63680b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
OTAViewController.m -[OTAViewController logWith:message:]:394 00 peripheral.writeValue(0x0000000000000000c0f80000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
OTAViewController.m -[OTAViewController logWith:message:]:394 00 [Callback] Central Manager did disconnect peripheral
OTAViewController.m -[OTAViewController logWith:message:]:394 05 Disconnected by the remote device
OTAViewController.m -[OTAViewController dfuError:didOccurWithMessage:]:343 error 202 Device disconnected unexpectedly
OTAViewController.m -[OTAViewController logWith:message:]:394 15 Upload aborted. Part 1 flashed sucessfully
```

I started the DFU like this:
```
[self setInitiator:[[DFUServiceInitiator alloc] initWithCentralManager:_cbCentralManager target:_discoveredPeripheral]];
[_initiator withFirmware:firmware];
_initiator.packetReceiptNotificationParameter = 15;
_initiator.logger = self;
_initiator.delegate = self;
_initiator.progressDelegate = self;
[self setDfuServiceController:[[_initiator start] retain]];
```

I have the same problem when I use the nRF Toolbox from iOS. However, I am able to flash it with an android phone that is using [Android-DFU-Library](https://github.com/NordicSemiconductor/Android-DFU-Library) version 1.1.1 and a packet receipt notification parameter of 15.

mostafaberg, 2016-12-13:
Can you try setting PRN to 12 instead of 15 and report back ? it's a bit different on iOS than android

philips77, 2016-12-13:
Hi @starblanket,
I see 3 things:
1. You are sending 2 parts, SD and/or BL and then App, Am I correct? And the issue you get is during the second connection.
2. Do you call abort() method? The last log message says the DFU was aborted so the abort flag must have been set somewhere.
3. Also looks like you are using quite old version of DFU bootloader, from SDK before 7.0, where there was no DFU Version characteristic.

And here are my conclusions. The worst is that they all exclude each other.
1. First of all why do abort the DFU? Is it intentional? If this message shows up there without you calling ```controller.abort()``` that's a bug.
2. If you are using the old DFU bootloader the library should find out whether you are in app mode or in bootloader mode by counting services. However, I think I made a bug, and I discover only DFU-related services, so always the number of services will be 1. In your case it should assume you are in app mode when you have more than 1 service, and in bootloader mode when only DFU service was discovered. If your design can perform DFU while in app mode, you have to set the *forceDfu* flag *in _initiator* to true. I will fix this issue soon, no matter you have it or not.
3. Did you flash SD/BL in the first connection successfully and failed on the second? If not, the last log message is not correct.

philips77, 2016-12-13:
As for the last log message, I think you call abort() in the ```[OTAViewController dfuError:didOccurWithMessage:]``` method. You shouldn't. DFU was already terminated and destroyed at that time.

philips77, 2016-12-13:
Hi, I did a commit https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/commit/a7788dead8eda7b383f64d8c00bc84363e551318 which should help you, if your DFU bootloader is from SDK 6.0 or 6.1 (no DFU Version characteristic, Buttonless Jump supported). It's on *develop* branch, so to test if you would have to modify your Podfile like this: https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/master/Podfile#L3 (but set branch name to **develop** and then type ```pod update```.
Could you try it and say if it works?

starblanket, 2016-12-13:
Using a PRN of 12 or 15 makes no difference. I tried again with nRF Toolbox (4.2.0) and confirmed it also does not work:

```
Verbose: Connecting to MyDevice...
Debug: centralManager.connect(peripheral, options:nil)
Debug: [Callback] Central Manager did connect peripheral
Info: Connected to MyDevice
Verbose: Discovering services...
Debug: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123, FE59])
Info: Services discovered
Info: Connected to MyDevice
Verbose: Legacy DFU Service found
Verbose: Discovering characteristics in DFU Service...
Debug: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
Info: DFU characteristics discovered
Verbose: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Verbose: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
Application: DFU Control Point notifications enabled
Verbose: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Verbose: Writing image sizes (0b, 0b, 63680b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
Debug: peripheral.writeValue(0x0000000000000000c0f80000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
Debug: [Callback] Central Manager did disconnect peripheral
Info: Disconnected by the remote device
```

starblanket, 2016-12-13:
Thanks @philips77. The device was built with mbed. `BLE::getVersion()` returns "Nordic BLE4.1 ver:8 fw:0067" (which is S130 v1.0.0). It sounds like your patch may not apply, but I will give it try and let you know.

The zip file I am flashing contains only an app. You are seeing the complete log from immediately after `[self setDfuServiceController:[[_initiator start] retain]];` was invoked.

Yes, I had been calling abort() in `[OTAViewController dfuError:didOccurWithMessage:]`, but I will remove that. Changing that has no impact on this problem however.

philips77, 2016-12-13:
I was talking about the DFU Version characteristic, not BLE::getVersion(). This char. may return 0x0100 for app mode, or one of 0x0500, 0x0600, 0x0700, 0x0800 in bootloader mode.
If there is no such app the library will count services, as I wrote.
It looks like you are in the app mode, you don't have DFU Version char (otherwise reading it would be visible in the log), and sending 0x0104 will trigger switching to DFU bootloader mode. The behavior is correct, but the lib does not expect disconnection as it should. Please, try with the change.

starblanket, 2016-12-13:
There is no DFU Revision Characteristic (UUID 0x1534) in the Device Firmware Update Service. All I see are the DFU Control Point and DFU Packet characteristics.

I am expecting the DFU library to switch to bootloader mode and it does do that. But then disconnects. I tested with this pod:

`pod 'iOSDFULibrary', :git => 'https://github.com/nordicsemiconductor/ios-pods-dfu-library', :branch => 'develop'`

and the behavior did not change. I still get the disconnect at the same place.

philips77, 2016-12-13:
Hmm.. My fix fixed a case when the library would assume that the peripheral is in app mode and should jump to bootloader, not the other way around.
In your case it thinks it's in bootloader mode while in fact is in app mode. Do you have any other services on your device when in app mode except dfu?
Btw, are you sure you have Service Changed characteristic in both app and dfu mode?

starblanket, 2016-12-13:
The device is in app mode before ` [_initiator start]`, but nothing is connected to it. I do have a custom service when in app mode. I have the LEDs on the device flashing different patterns when in app mode and DFU mode. I do see the expected pattern when DFU starts. So it is jumping to the bootloader.

I used the nRF Connect application to confirm that the Generic Attribute Service Changed characteristic exists both in app and DFU modes. This property is set to "indicate".

Hmmm. My view controller is such that I can re-try the flashing after the this first disconnect, while the device is already at bootloader and before the bootloader times out and returns to app mode. **That second try works** when the device is sitting at the bootloader when [_initiator start] was invoked. So maybe this is just a side effect of [Issue 62](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/62). If the library just re-tried, it looks like all would be fine.

philips77, 2016-12-15:
Hi, we've just released version 3.0.1 to CocoaPods.
Now the lib discovers all services after it connects to a peripheral, so if there is no DFU Version characteristic inside the DFU Service it should estimate the mode based on number of services. As you have a custom service it should know that it is now in app mode and should handle the disconnect event correctly. I hope.
Could you try and post logs again if it fails?

starblanket, 2016-12-15:
The outward behavior did not change with 3.0.1. I still get the disconnect:

```
-[OTAViewController installFirmware]:166 parts=1 softdevice=0 bootloader=0 application=63680
-[OTAViewController logWith:message:]:393 01 Connecting to MyDevice...
-[OTAViewController logWith:message:]:393 00 centralManager.connect(peripheral, options:nil)
-[OTAViewController dfuStateDidChangeTo:]:269 DFU state change to 0
-[OTAViewController logWith:message:]:393 00 [Callback] Central Manager did connect peripheral
-[OTAViewController logWith:message:]:393 05 Connected to MyDevice
-[OTAViewController logWith:message:]:393 01 Discovering services...
-[OTAViewController logWith:message:]:393 00 peripheral.discoverServices(nil)
-[OTAViewController logWith:message:]:393 05 Services discovered
-[OTAViewController logWith:message:]:393 05 Connected to MyDevice
-[OTAViewController logWith:message:]:393 01 Legacy DFU Service found
-[OTAViewController logWith:message:]:393 01 Discovering characteristics in DFU Service...
-[OTAViewController logWith:message:]:393 00 peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
-[OTAViewController logWith:message:]:393 05 DFU characteristics discovered
-[OTAViewController logWith:message:]:393 01 Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
-[OTAViewController dfuStateDidChangeTo:]:269 DFU state change to 1
-[OTAViewController logWith:message:]:393 01 Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:393 10 DFU Control Point notifications enabled
-[OTAViewController logWith:message:]:393 01 Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
-[OTAViewController logWith:message:]:393 01 Writing image sizes (0b, 0b, 63680b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.writeValue(0x0000000000000000c0f80000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
-[OTAViewController logWith:message:]:393 00 [Callback] Central Manager did disconnect peripheral
-[OTAViewController logWith:message:]:393 05 Disconnected by the remote device
-[OTAViewController dfuError:didOccurWithMessage:]:341 error 202 Device disconnected unexpectedly
```

philips77, 2016-12-15:
Are you sure you have do not set ```initiator.forceDfu = true```? It shold be ```false``` in your case.

philips77, 2016-12-15:
Here's my log output:

```
[...]
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Changed state to: Starting
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
W: Application with buttonless update found
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Changed state to: Enabling DFU Mode
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
```

I did the test on a peripheral with DFU Service without DFU Version characteristic and with an additional service. forceDfu is false. 2 services has been found, so [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift#L160) true is returned. Could you put your breakpoint here and check what's going on?

Also, remember that iOS does not expose Generic Access, Generic Attribute and HID services (and perhaps some more) so if the 'custom' service in your case is one of the protected ones it will not be returned during service discovery.

starblanket, 2016-12-15:
@phillips77 you are correct. I had forceDfu = true (by accident). I was fiddling with it yesterday and forgot I had it in there. It now works with forceDfu = false:

```
-[OTAViewController installFirmware]:166 parts=1 softdevice=0 bootloader=0 application=63680
-[OTAViewController logWith:message:]:393 01 Connecting to MyDevice...
-[OTAViewController logWith:message:]:393 00 centralManager.connect(peripheral, options:nil)
-[OTAViewController dfuStateDidChangeTo:]:269 DFU state change to 0
-[OTAViewController logWith:message:]:393 00 [Callback] Central Manager did connect peripheral
-[OTAViewController logWith:message:]:393 05 Connected to MyDevice
-[OTAViewController logWith:message:]:393 01 Discovering services...
-[OTAViewController logWith:message:]:393 00 peripheral.discoverServices(nil)
-[OTAViewController logWith:message:]:393 05 Services discovered
-[OTAViewController logWith:message:]:393 05 Connected to MyDevice
-[OTAViewController logWith:message:]:393 01 Legacy DFU Service found
-[OTAViewController logWith:message:]:393 01 Discovering characteristics in DFU Service...
-[OTAViewController logWith:message:]:393 00 peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
-[OTAViewController logWith:message:]:393 05 DFU characteristics discovered
-[OTAViewController logWith:message:]:393 01 Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
-[OTAViewController dfuStateDidChangeTo:]:269 DFU state change to 1
-[OTAViewController logWith:message:]:393 01 Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:393 10 DFU Control Point notifications enabled
-[OTAViewController logWith:message:]:393 15 Application with buttonless update found
-[OTAViewController logWith:message:]:393 01 Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
-[OTAViewController dfuStateDidChangeTo:]:269 DFU state change to 2
-[OTAViewController logWith:message:]:393 00 [Callback] Central Manager did disconnect peripheral
-[OTAViewController logWith:message:]:393 05 Disconnected by the remote device
-[OTAViewController logWith:message:]:393 01 Connecting to MyDevice...
-[OTAViewController logWith:message:]:393 00 centralManager.connect(peripheral, options:nil)
-[OTAViewController logWith:message:]:393 00 [Callback] Central Manager did connect peripheral
-[OTAViewController logWith:message:]:393 05 Connected to MyDevice
-[OTAViewController logWith:message:]:393 01 Discovering services...
-[OTAViewController logWith:message:]:393 00 peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
-[OTAViewController logWith:message:]:393 05 Services discovered
-[OTAViewController logWith:message:]:393 01 Legacy DFU Service found
-[OTAViewController logWith:message:]:393 01 Discovering characteristics in DFU Service...
-[OTAViewController logWith:message:]:393 00 peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
-[OTAViewController logWith:message:]:393 05 DFU characteristics discovered
-[OTAViewController logWith:message:]:393 01 Reading DFU Version number...
-[OTAViewController logWith:message:]:393 00 peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
-[OTAViewController logWith:message:]:393 05 Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0400
-[OTAViewController logWith:message:]:393 10 Version number read: 0.4
-[OTAViewController logWith:message:]:393 01 Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
-[OTAViewController dfuStateDidChangeTo:]:269 DFU state change to 1
-[OTAViewController logWith:message:]:393 01 Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:393 10 DFU Control Point notifications enabled
-[OTAViewController logWith:message:]:393 01 Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
-[OTAViewController logWith:message:]:393 01 Writing image sizes (0b, 0b, 63680b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
-[OTAViewController logWith:message:]:393 00 peripheral.writeValue(0x0000000000000000c0f80000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
-[OTAViewController logWith:message:]:393 05 Data written to 00001531-1212-EFDE-1523-785FEABCD123
-[OTAViewController logWith:message:]:393 10 Start DFU (Op Code = 1, Upload Mode = 4) request sent
-[OTAViewController logWith:message:]:393 05 Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
-[OTAViewController logWith:message:]:393 10 Response (Op Code = 1, Status = 1) received
-[OTAViewController logWith:message:]:393 10 Writing Initialize DFU Parameters...
etc.
```

I'm closing this. Thanks!

closed 2016-12-15 by starblanket