nordicsemi/IOS-DFU-Library #64: Crash if DFU Characteristic isn't found

casey-pax, 2016-12-13:
If the didDiscoverCharacteristicsFor service callback gets called with an error, it results in a nil unwrapping.

`#1	0x00000001008e6368 in LegacyDFUService.sendReset(onError : (DFUError, String) -> ()) -> () at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift:445
#2	0x00000001008df668 in LegacyDFUPeripheral.resetDevice() -> () at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Peripherals/LegacyDFUPeripheral.swift:179
#3	0x00000001008c1a18 in protocol witness for BaseDFUPeripheralAPI.resetDevice() -> () in conformance <A where ...> BaseDFUPeripheral<A> ()
#4	0x00000001008b2c20 in BaseDFUExecutor.error(DFUError, didOccurWithMessage : String) -> () at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUExecutor.swift:117
#5	0x00000001008dcef8 in protocol witness for BasePeripheralDelegate.error(DFUError, didOccurWithMessage : String) -> () in conformance LegacyDFUExecutor ()
#6	0x00000001008bc66c in BaseDFUPeripheral.(defaultErrorCallback.getter).(closure #1) at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift:76
#7	0x00000001008aa90c in thunk ()
#8	0x00000001008e6be0 in LegacyDFUService.peripheral(CBPeripheral, didDiscoverCharacteristicsFor : CBService, error : Error?) -> () at /Volumes/Development/pax-app-ios/PaxApp/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift:479
`

This is hard to reproduce, you probably have to lose connection at just the right spot.

philips77, 2016-12-13:
Oh, sorry. I'll look into it tomorrow.

philips77, 2016-12-15:
Fixed in version 3.0.1. Thanks again for finding it!

closed 2016-12-15 by philips77