nordicsemi/IOS-DFU-Library #65: BT device advertises only DFU service after firmware update.

anlaital, 2016-12-15:
Hi,

We are using the `3.0.1` version of the SDK. On iOS 10.2 there seems to be some functional change in the BT stack, as our device gets stuck advertising only DFU service `00001530-1212-EFDE-1523-785FEABCD123` after firmware update. This issue does not happen on previous iOS releases. The only fix is to either reboot the device or to turn Bluetooth on the device off and then back on. Restarting the app doesn't help at all, even after force-quitting.

Do you have any idea what could cause this and any suggestions on how to fix it? This is most likely caused by something that we do incorrectly.

philips77, 2016-12-15:
Hi, I don't understand fully. Could you give some more details. When your devices advertises with this UUID is it in dfu mode? Did the DFU upload complete? Does is stuck after switching from app mode to DFU mode and can't connect or at the end? 
I tried today DFU on iPhone 7 with 10.2 and it send to work. Passed all my tests.

anlaital, 2016-12-15:
DFU upload completes and the firmware is updated. The fact that this issue is fixed by turning off bluetooth on the device and then back on seems to indicate that causes either the device to remain in DFU mode or the broadcasted services not updating correctly.

Do you have your DFU-related test codes in this repository as well?

philips77, 2016-12-15:
You are turning Bluetooth off an on on the device that you have updated our on the phone? After successful DFU the target device should reboot into new app mode. If it doesn't it something wrong with the firmware, I guess. 
The phone disconnects successfully of remains connected and turning off Bluetooth disconnects it?

philips77, 2016-12-15:
Unfortunately we are lacking any unit tests despite our good will, but we accept pull requests :) I'm just performing tests with different scenarios manually.

anlaital, 2016-12-15:
I am turning the bluetooth off and back on again on the iPhone, not the peripheral. I think this is not a problem on the peripheral or firmware side, as doing this solely on the phone side fixes the issue. This would also explain why the issue occurs only on iOS 10.2 and no other iOS version so far.

Recreating the CBCentralManager after DFU results in `CBCentralManagerStateUnsupported` being notified in the `centralManagerDidUpdateState` callback.

I don't think there's anything wrong with your SDK, but I would appreciate any help or hints you could give me on how to debug this issue. :)

Do you have example test code that connects to a peripheral, puts it into DFU mode, does the firmware update and then reconnects to it that works on iOS 10.2?

philips77, 2016-12-15:
We are using nRF Toolbox for that and are working on adding dfu to nRF Connect on iOS.
For development and testing we use the example project.

philips77, 2016-12-15:
I'll do some more test tomorrow with 10.2 and try to reproduce your issue. What version of DFU are you using?

anlaital, 2016-12-15:
Could you elaborate on what you mean by the `version of DFU`?

philips77, 2016-12-16:
What Nordic SDK did you branch your project from? There were some changes in DFU between them, new features were added etc.

anlaital, 2016-12-16:
Quick update on this issue. It seems that the way how iOS caches ble things has changed in 10.2. Our peripheral's DFU mode did not advertise the app service, so it got cached by iOS was not found even after switching out of the DFU mode.

This could be fixed by waiting for the cache TTL expiration (~15 mins) or by turning the bluetooth off and on again on the phone.

The actual fix we did was to advertise both services also in DFU mode.

The SDK we are using is `3.0.0`.

anlaital, 2016-12-16:
@philips77 there seems to be regression in the SDK version `3.0.1` as it gets stuck in waiting for DFU mode confirmation. This works correctly with SDK version `3.0.0`.

philips77, 2016-12-16:
I was asking of the SDK version that you have based your firmware on, not iOS lib version.
Regarding the last comment. could you paste logs you get with loggerDelegate here?

anlaital, 2016-12-16:
My apologies. `nRF51SDK` version is `6.1.0` and `SoftDevice S110` version is `v7.3.0` on the firmware side. I will give you the delegate logs here shortly. 👍

anlaital, 2016-12-16:
The issue seems to have vanished after I rolled back to `3.0.0`, tested and then reinstalled `3.0.1`. Both firmware and bootloader updates are successful. I will open another issue should this occur again, but for now, let's close this.

Thank you @philips77 for your help!

closed 2016-12-16 by anlaital