nordicsemi/IOS-DFU-Library #66: I   was crash in the library when  I  use the library update the firmware

duanshixuan, 2016-12-16:
the reason is 
 *** Terminating app due to uncaught exception 'NSRangeException', reason: '*** -[_NSInlineData getBytes:range:]: range {2, 1} exceeds data length 2'
*** First throw call stack:
(0x18de441c0 0x18c87c55c 0x18de44108 0x18e84644c 0x1007259c0 0x1004f1420 0x1004f59cc 0x1004f6a50 0x194a1a364 0x194a1a4c4 0x194a16ef4 0x194a12018 0x194a25f58 0x194a1e70c 0x1008cd25c 0x1008cd21c 0x1008dab54 0x1008d0ce4 0x1008d2060 0x18ddf1f2c 0x18ddefb18 0x18dd1e048 0x18f7a1198 0x193cf7818 0x193cf2550 0x100149644 0x18cd005b8)
libc++abi.dylib: terminating with uncaught exception of type NSException

this is my code : 
 DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[NSURL fileURLWithPath:filePath] ];
                            NSLog(@"filename:%@",selectedFirmware.fileName);
                            if (selectedFirmware.fileName) {
                                DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:vc.heartRateCentralManager target:self.per.per];
                                
                                // Optional:
                                initiator.forceDfu = NO; // default NO
                                initiator.packetReceiptNotificationParameter = 12; // default is 12
                                initiator.logger = self; // - to get log info
                                initiator.delegate = self; // - to be informed about current state and errors
                                initiator.progressDelegate = self; // - to show progress bar
//                                initiator.peripheralSelector = ... // the default selector is used
                                DFUServiceInitiator *initiator1 = [initiator withFirmware:selectedFirmware];
                                self.dfuServiceController = [initiator1 start];
                            }

philips77, 2016-12-16:
Hello,
Why do you create another copy of initiator? Try like this:
```
DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[NSURL fileURLWithPath:filePath] ];
NSLog(@"filename:%@",selectedFirmware.fileName);
if (selectedFirmware.fileName) {
   DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:vc.heartRateCentralManager target:self.per.per];

   // Optional:
   initiator.forceDfu = NO; // default NO
   initiator.packetReceiptNotificationParameter = 12; // default is 12
   initiator.logger = self; // - to get log info
   initiator.delegate = self; // - to be informed about current state and errors
   initiator.progressDelegate = self; // - to show progress bar

   // initiator.peripheralSelector = ... // the default selector is used
   self.dfuServiceController = [[initiator withFirmware:selectedFirmware] start];
}
```

philips77, 2016-12-16:
Also, what version of the DFU library are you using? The current one is 3.0.1.

duanshixuan, 2016-12-18:
The version of the DFU library  is 3.0.1.   I follow your approach to do, or the same reason for the error

duanshixuan, 2016-12-18:
The crash is located on line 157 in the DFUControlPoint.swift

philips77, 2016-12-18:
Why the response you get is 2, not 3, bytes long?
In Legacy dfu it should always contain 0x10-Req-Status, as in the code. In what moment so you get it and what version of dfu do you use (SDK version)?
If you have a custom DFU Bootloader, you would have to modify the lib on your own. But it's not recommended as you will have more work in case of some changes or bug fixing we may do.

duanshixuan, 2016-12-18:
I can update firmware sucessfully with the example application from IOS-Pods-DFU-Library. But it didn't work, when I try to update the same firmware to the same device with Object-C. Do you have any example application for Object-C?

philips77, 2016-12-18:
It doesn't matter which language you use in your project. The code I posted above is enough.
If your upload crashes at line 157 it means that your peripheral sends 2 byte response, instead of 3. Investigate that. I guess it happens only sometimes and even with the example app.

duanshixuan, 2016-12-19:
DFU version is 3.0.1  ,SDK version is nRF5_SDK_11.0.0 .  we use the standard dfu bootloader from Nordic .

duanshixuan, 2016-12-19:
What's meaning of the 0x10-Req-Status in the Legacy dfu?  
When I launch the dfu application and start download the firmware image, but  the peripheral  receives no message.

duanshixuan, 2016-12-19:
I'm not sure if I have transfer the peripheral device info into the duf library correctly? 
The  example app  works  fine with the peripheral.

philips77, 2016-12-19:
Could you post here logs that you get with logger delegate in your app?
>What's meaning of the 0x10-Req-Status in the Legacy dfu?

After a command has been sent to the DFU target (which consist of: [RequestCode-SomeAdditionalBytes] a response is then sent: [ResponseCode=0x10-RequestCode-Status]. All values are 1-byte long so you always should get 3 bytes, not 2. So I would like to see the reason why you receive 2 bytes there.

duanshixuan, 2016-12-22:
Hi,philips,
here is the log for my issue. Would you please help me out?

2016-12-22 17:31:32.345 FizzoMark[1494:573686] I: Connected to FIZZO_GJ37
2016-12-22 17:31:32.348 FizzoMark[1494:573686] I: Connected to FIZZO_GJ37
2016-12-22 17:31:32.349 FizzoMark[1494:573686] V: Legacy DFU Service found
2016-12-22 17:31:32.349 FizzoMark[1494:573686] V: Discovering characteristics in DFU Service...
2016-12-22 17:31:32.349 FizzoMark[1494:573686] D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
2016-12-22 17:31:32.353 FizzoMark[1494:573686] I: DFU characteristics discovered
2016-12-22 17:31:32.354 FizzoMark[1494:573686] V: Reading DFU Version number...
2016-12-22 17:31:32.354 FizzoMark[1494:573686] D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
2016-12-22 17:31:32.386 FizzoMark[1494:573686] I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
2016-12-22 17:31:32.386 FizzoMark[1494:573686] A: Version number read: 0.1 
2016-12-22 17:31:32.387 FizzoMark[1494:573686] V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
2016-12-22 17:31:32.387 FizzoMark[1494:573686] D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
2016-12-22 17:31:32.475 FizzoMark[1494:573686] I: Notification received from 2A19, value (0x): 1b
2016-12-22 17:32:02.898 FizzoMark[1494:573686] *** Terminating app due to uncaught exception 'NSRangeException', reason: '*** -[_NSInlineData getBytes:range:]: range {1, 1} exceeds data length 1'
*** First throw call stack:
(0x18651422c 0x1981e00e4 0x18651416c 0x1873484d4 0x1006b0914 0x1003d50c0 0x1003d96c0 0x1003da744 0x186180b3c 0x18617c9d4 0x1008b4fd4 0x1008b4f94 0x1008bfdb8 0x1008b82c4 0x1008b96f4 0x1864cb544 0x1864c95ec 0x1863f4f74 0x18fe4f6fc 0x18aff6d94 0x1000e92cc 0x19888aa08)
libc++abi.dylib: terminating with uncaught exception of type NSException

philips77, 2016-12-22:
Hi, thanks for the log!
As you see, the crash is caused by a notification that should completely be ignored as it comes from Battery Level characteristic (UUID: 2A19), nothing related to DFU.
I did not expect other notifications to be sent during DFU, but as the library works with already connected devices, or with bonded devices that might have CCCD enabled before - why not.
I will fix your issue right away and publish version 3.0.2. Please, give me an hour.

philips77, 2016-12-22:
Fixed in 3.0.3 (actually in 3.02, but I made 3.0.3 moment later with licence fix).

closed 2016-12-22 by philips77

duanshixuan, 2016-12-22:
It Works!!! Thanks!