nordicsemi/IOS-DFU-Library #67: Error pod 'iOSDFULibrary','~> 3.0.1'

AnD010, 2016-12-16:
Hello!
 I'm trying to use the Swift library in my objective c project.
It's the first time I use Swift, I do not know if I'm doing something wrong.
My podfile is 
```
use_frameworks!
source 'https:......PRIVATE-FRAMEWORK'
source 'https://github.com/CocoaPods/Specs.git'
target 'TestWork' do
pod  'TSMessages'
pod 'skpsmtpmessage'
pod 'PrivateFramework','~> 1.5.1'
pod 'iOSDFULibrary','~> 3.0.1'
end
```
And the console result is 

```
> [!] Unable to satisfy the following requirements:
> 
> - `iOSDFULibrary (~> 3.0.1)` required by `Podfile`
> 
> None of your spec sources contain a spec satisfying the dependency: `iOSDFULibrary (~> 3.0.1)`.
> 
> You have either:
>  * out-of-date source repos which you can update with `pod repo update`.
>  * mistyped the name or version.
>  * not added the source repo that hosts the Podspec to your Podfile.
> 
> Note: as of CocoaPods 1.0, `pod repo update` does not happen on `pod install` by default.
```
If I try 
pod 'iOSDFULibrary'
I get as a result 
`Installing iOSDFULibrary (1.0.12)` 
Then I have a lot of problem about Swift version, xcode 8 said I need to pass to new Swift version 2.0 and I have many errors.

thanks!

philips77, 2016-12-16:
Perhaps it's something related to ```source ...```?
Check this: https://cocoapods.org/?q=iOSDF
@mostafaberg ?

AnD010, 2016-12-16:
Thanks for replying @philips77 , but if you mean
```
Source 'https: ...... PRIVATE-FRAMEWORK'
Source 'https://github.com/CocoaPods/Specs.git'
```
The first is to access a private repository, and the second is to access the cocoa public repository.
And if I change pod 'iOSDFULibrary','~> 3.0.1' to pod 'iOSDFULibrary', the result is correct, but I install this version  Installing iOSDFULibrary (1.0.12) in which I have problems about Swift version, xcode 8 said I need to pass to new Swift version 3.0 and I have a lot of errors.

thanks

philips77, 2016-12-16:
Hm.. I found version 3.0.1 on this repo: https://github.com/CocoaPods/Specs/tree/master/Specs/0/e/5/iOSDFULibrary. Do you have cocoapods up to date?

mostafaberg, 2016-12-16:
You don't need to have `source xxx` there at all. just use the proper `pod` line instead.
follow what the command tells you here:
```
>  * out-of-date source repos which you can update with `pod repo update`.
```

after the repo updates, run `pod update`,not `pod install`, simply installing will fetch the version in the lock file

mostafaberg, 2016-12-16:
Also, don't lock it to `3.0.1`, `3.0` would be better i'd say so you get bugfixes as soon as they're out

Your Podfile should look like this:
```
use_frameworks!
source 'https:......PRIVATE-FRAMEWORK'
target 'TestWork' do
    pod  'TSMessages'
    pod 'skpsmtpmessage'
    pod 'PrivateFramework', '~> 1.5.1'
    pod 'iOSDFULibrary', '~> 3.0'
end
```

AnD010, 2016-12-16:
Thanks @philips77  and @mostafaberg , I did pod update and now 
```
Downloading dependencies
Using PrivateFramework (1.5.1)
Using HexColors (2.2.1)
Using TSMessages (0.9.12)
Using Zip (0.6.0)
Installing iOSDFULibrary 3.0.1 (was 1.0.12)
Using skpsmtpmessage (0.0.1)
```
But I honestly I don´t understand.
I see in your iOSDFULibrary.podspec
```
S.version = "3.0.1"
S.source = {: git => "https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library.git" ,: tag => s.version.to_s}
```
And in the link that send to me @ philips77, the tag "3.0.1" ...
Maybe it is wrong, but whenever I have modified my PrivateFramework, I have created the tag 
and I did a pod install and everything worked fine for me...And with other repositories too...
My bad...
Thanks for your time guys!