nordicsemi/IOS-DFU-Library #69: Peripheral fails to respond to jump to bootloader command.

anlaital, 2016-12-22:
Hi,

I'm opening a new issue for this as this was previously discussed briefly in #65.

Using the `3.0.1` version of the library fails to get response from the peripheral after sending the `.jumpToBootloader` command. This works correctly on the `3.0.0` version of the library, so there had to be some regression between those releases. The changeset is fortunately very minimal, so finding the culprit was quite easy.

https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/compare/3.0.0...3.0.1#diff-b906bc607dacb47dda5a8f7ff1a77068L64

Using the `3.0.1` library version and reverting this change fixes the issue for us. However, I don't fully understand why this causes issues for us. If you have any ideas, I'm all ears. :)

anlaital, 2017-01-05:
@philips77 might you have any idea as to why this happens?

philips77, 2017-01-05:
Hi @anlaital,
we have to discover all services for backwards compatibility. Based on number of services we can reason whether we are in DFU mode or in App mode when connected to devices based on SDKs below 7.
Why does it make a difference for you?

anlaital, 2017-01-05:
That is an excellent question. I don't know why it makes a difference, but by reverting this change we can update the firmware again. If this change is present, the DFU process never receives any data from the peripheral. This means that `DFUControlPoint`'s `func peripheral(_ peripheral: CBPeripheral, didUpdateValueFor characteristic: CBCharacteristic, error: Error?)` callback is never called.

philips77, 2017-01-10:
Could you send us any HEX with fw that we could run on the DK where we could investigate this issue? It seems for me very strange that when you discover all characteristics you don't get updates, but when only one you do. I tried to come up with an idea, but no luck.
Are you sure you are not getting a different update/change peripheral.delegate when you discover another service?

mostafaberg, 2017-01-24:
Any updates here ? will close later in the day if we don't hear back assuming you got things working :), else, please tell us the status of the issue so we can take a look at anything else

anlaital, 2017-01-24:
We're going about rewriting the BT stack in our app in hopes that it would fix this issue also. We can reopen this issue if it still happens in a separated test application. This will happen sometime in February, so let's close this issue for now.

closed 2017-01-24 by anlaital

mostafaberg, 2017-01-24:
Perfect !, let us know if things go ok or if you encounter further issues, thanks