nordicsemi/IOS-DFU-Library #70: upload appstore failed,  contains disallowed file

JJCoderiOS, 2016-12-23:
hi  , When I uploaded to AppStore failed, show the following information,
i use the latest Xcode version  
[image]

JJCoderiOS, 2016-12-23:
i read this art is not  success   http://stackoverflow.com/questions/30547283/submit-to-app-store-issues

mostafaberg, 2016-12-23:
Could be an issue with Cocoapods or your project configuration.
-Which version of cocoapods are you using ?
-How are you building the project ?

JJCoderiOS, 2016-12-23:
hi,i have solved the problem ,i run  script :

cd "${CONFIGURATION_BUILD_DIR}/${UNLOCALIZED_RESOURCES_FOLDER_PATH}/"
if [[ -d "Frameworks" ]]; then
rm -fr Frameworks
fi

in build phases 

however, the ZIP and iOSDFULibrary  framework  make the ipa huge....

mostafaberg, 2016-12-23:
This is a workaround not a fix, can you please answer my questions ?

JJCoderiOS, 2016-12-23:
cocoapods version   1.1.0.beta.2
how to buildiong the project ?  what's your means?

mostafaberg, 2016-12-23:
Prefect, when i mean building i mean what are the steps that you do in Xcode to get the IPA file ?

JJCoderiOS, 2016-12-23:
aha~~~~  Product - > archive ~~~

mostafaberg, 2016-12-23:
That sounds about right, could it be something with your build settings ? what are you `valid architectures` values set to ? are you targetting MacOS ? or just iOS ?

mostafaberg, 2017-01-10:
@JJCoderiOS Any news ? did you get your issue resolved ? This issue will be closed in 24 hours if we got no response and we will assume you fixed it

closed 2017-01-12 by philips77