nordicsemi/IOS-DFU-Library #71: DFUController.restart() method has been removed – how can I restart the DFU process?

larsblumberg, 2017-01-02:
In earlier versions of this library (at least 1.x) there was a method `restart()` for protocol `DFUController`. It was very useful, because very often the first firmware update fails with error code `DFUError.deviceDisconnected`.

What do you suggest ist the best way to recover from a `DFUError.deviceDisconnected` during a firmware update?

(This issue might be related to #62)

mostafaberg, 2017-01-02:
for Secure DFU there's nothing special to be done, simply starting the DFU process again will automatically resume from the last point

larsblumberg, 2017-01-02:
What about non-secure updates? We're using an older Nordic SDK and thus can't provide secure updates yet. This said, "legacy updates" are performed.

philips77, 2017-01-02:
In legacy DFU the device should be automatically resetted when DFU fails or is aborted. No need to do anything.

philips77, 2017-01-02:
There is actually more problems with Secure DFU, as the current impl (SDK 12.x) does not support resetting back to app mode, as far as i know.

larsblumberg, 2017-01-02:
Why shouldn't we do anything? What about the user not getting his firmware update?

philips77, 2017-01-02:
What do you mean by " user not getting his firmware update"? Could you elaborate?

Whenever an error occurs, the resetDevice() method is called: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUExecutor.swift#L113 
For Legacy DFU it does this: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/LegacyDFU/Peripherals/LegacyDFUPeripheral.swift#L177

larsblumberg, 2017-01-02:
If the firmware update has failed, the user's device will still carry the old firmware version.

How can we provide an automatic update retry mechanism from the app side? In older versions of this library we would simply call `restart()` – please see my original question on the top.

philips77, 2017-01-02:
Ok, I get it now! :) Sorry!

philips77, 2017-01-02:
I'll have a look at the code and try to add it back. Sorry for that. Never thought someone was using it actually :)

larsblumberg, 2017-01-02:
It turns out, a firmware update retry can be implemented as following. I argue that your library should provide this method again as it was already party if older versions. I want stress again that it is a bad user experience if something fails and we don't retry, especially in this case where it solves the issue as easy that.

```
let dfuInitiator = DFUServiceInitiator(...)
...
var dfuController = dfuInitiator.start()
...
public func dfuError(_ error: DFUError, didOccurWithMessage message: String) {
    if error == .deviceDisconnected {
        dfuController.abort()
        dfuController = dfuInitiator.start()
    }
}
```

Again, it would be great if I could simply call `dfuController.restart()` instead of the workaround above.

Anyway, with that workaround we can again provide a better user experience. Failed firmware updates happen very often on iOS and macOS. I guess there's some bug in this library?

philips77, 2017-01-02:
> Failed firmware updates happen very often on iOS and macOS. I guess there's some bug in this library?
I actually find the lib very stable, I have to say, but we are using just the simple HRM apps for testing. I will look into the ```restart()``` and retry problem.
Could you tell when do you get most errors? And what kind of errors?

larsblumberg, 2017-01-02:
The lib _is_ stable, except the fact the firmware updates nearly never work on the first try. As already mentioned above `DFUError.deviceDisconnected` is being sent from the library via `DFUServiceDelegate` when doing legacy updates.

larsblumberg, 2017-01-02:
Do you test with legacy firmware Nordic SDK version 8.0, too?

philips77, 2017-01-02:
Yes, we do. I'll come back to you tomorrow.

philips77, 2017-01-05:
Hi, I've added the restart() method in Version3 branch (https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/tree/Version3). Could you give it a try before I push to Pods?

> Modify the Podile like here: https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/master/Podfile#L3

philips77, 2017-01-10:
Hi @larsblumberg , did you test the new/old method and the latest changes on Version3 branch? Should we release it on Pods?

larsblumberg, 2017-01-10:
Sorry for my late reply. I didn't yet test the branch since I'm using the workaround I mentioned above.

After you pushed a new DFU library version I will check.

philips77, 2017-01-12:
Hi, the version 3.0.4 is now available on Pods. I'm closing this issue. Please reopen it if you find this feature not working as expected.

closed 2017-01-12 by philips77