nordicsemi/IOS-DFU-Library #72: Crash in DFUPeripheral line 161

larsblumberg, 2017-01-11:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L161

It just crashed for me here:

```
    func abort() -> Bool {
        aborted = true
        if peripheral?.state == .connecting { // <<<<<------- CRASH
            disconnect()
        }
        return true
    }
```

Reason is because you're force-unwrapping the peripheral. I've seen this in a few places in the library. It should be generally avoided.

philips77, 2017-01-11:
I'm forcing unwrapping when the variable needs to be set so the crash is as fast as possible. 
But here it definitely should be ? as peripheral may legally be nil when you scan for new one. Will change it right away.

larsblumberg, 2017-01-11:
Thanks a lot for fixing it so quickly.

philips77, 2017-01-11:
I pushed some commits. One fixes this issue, another just replace ifs with guards. Hope it still works.

closed 2017-01-11 by philips77